linux-can/can-utils #5: AutoGen.sh not found

droter, 2015-09-01:
Hi,

When I try to generate the auto config I am getting this message:
----------
autoreconf
----------

./autogen.sh: line 20: autoreconf: command not found
viki@c3po:~/Downloads/can-utils$ ./autogen.sh 
----------
autoreconf
----------
./autogen.sh: line 20: autoreconf: command not found

yegorich, 2015-09-02:
You are missing *autoconf* and *automake* packages. What Linux distro are you using?

pfreitasaf, 2015-11-26:
I am having the same problem using ubuntu.

marckleinebudde, 2015-11-27:
Try installing these packages:

    sudo aptitude install automake autoconf

pfreitasaf, 2015-11-27:
Thanks, it solved that one problem, but now another appears:
GNUmakefile.am:25: error: Libtool library used but 'LIBTOOL' is undefined

marckleinebudde, 2015-11-27:
Try installing these packages:

    sudo aptitude install libtool

pfreitasaf, 2015-11-27:
Worked! Thanks!

closed 2015-11-27 by marckleinebudde

milad6506, 2015-12-23:
hi,
I tried all the instructions but fails in the make install process.

marckleinebudde, 2015-12-23:
For the record:
@milad6506 opened the issue https://github.com/linux-can/can-utils/issues/12 to discuss this.