linux-can/can-utils #11: CANgw is not configured to work

shabbir-hussain, 2015-11-12:
CANgw tool gives error "netlink error -95 (Operation not supported)".

marckleinebudde, 2015-11-12:
Have you enabled the CAN_GW in you kernel, either as a module or compiled in? If it's a module, have you loaded it? Please give the complete command line of the cangw tool.

cheers, Marc

shabbir-hussain, 2015-11-12:
I'm using a BeagleBone Black with a can shield. I'm running the command "cangw -F" and it gives me the above error. Running modprobe gives the following "modprobe: FATAL: Module cangw not found"

How can I enable CAN_GW or load it as a module? Is there a usage doc?

marckleinebudde, 2015-11-12:
The module is called `can-gw`. In order to enable it, you have to recompile your kernel do a `make menuconfig` go to `Networking support` -> `CAN bus subsystem support` -> `CAN Gateway/Router (with netlink configuration)`.

hartkopp, 2016-01-16:
Seems to be solved.

closed 2016-01-16 by hartkopp

rubberduck203, 2018-03-30:
Before you go compiling the kernel, you probably want to try just loading the module.

```bash
sudo modprobe can-gw
```