linux-can/can-utils #12: installation fails

milad6506, 2015-12-23:
hi,
I am trying to install can-utils on Ubuntu 15.
when I insert the make install command it fails.
can anyone help me with this issue?

milad6506, 2015-12-23:
here is the message:
make[1]: Entering directory '/home/milad/Documents/can utils/can-utils-master'
 /bin/mkdir -p '/usr/local/bin'
  /bin/bash ./libtool   --mode=install /usr/bin/install -c asc2log bcmserver can-calc-bit-timing canbusload candump canfdtest cangen cangw canlogserver canplayer cansend cansniffer isotpdump isotpperf isotprecv isotpsend isotpserver isotpsniffer isotptun log2asc log2long slcan_attach slcand slcanpty '/usr/local/bin'
libtool: install: /usr/bin/install -c asc2log /usr/local/bin/asc2log
libtool: install: /usr/bin/install -c bcmserver /usr/local/bin/bcmserver
libtool: install: /usr/bin/install -c can-calc-bit-timing /usr/local/bin/can-calc-bit-timing
libtool: install: /usr/bin/install -c canbusload /usr/local/bin/canbusload
libtool: install: /usr/bin/install -c candump /usr/local/bin/candump
libtool: install: /usr/bin/install -c canfdtest /usr/local/bin/canfdtest
libtool: install: /usr/bin/install -c cangen /usr/local/bin/cangen
libtool: install: /usr/bin/install -c cangw /usr/local/bin/cangw
libtool: install: /usr/bin/install -c canlogserver /usr/local/bin/canlogserver
libtool: install: /usr/bin/install -c canplayer /usr/local/bin/canplayer
libtool: install: /usr/bin/install -c cansend /usr/local/bin/cansend
libtool: install: /usr/bin/install -c cansniffer /usr/local/bin/cansniffer
libtool: install: /usr/bin/install -c isotpdump /usr/local/bin/isotpdump
libtool: install: /usr/bin/install -c isotpperf /usr/local/bin/isotpperf
libtool: install: /usr/bin/install -c isotprecv /usr/local/bin/isotprecv
libtool: install: /usr/bin/install -c isotpsend /usr/local/bin/isotpsend
libtool: install: /usr/bin/install -c isotpserver /usr/local/bin/isotpserver
libtool: install: /usr/bin/install -c isotpsniffer /usr/local/bin/isotpsniffer
libtool: install: /usr/bin/install -c isotptun /usr/local/bin/isotptun
libtool: install: /usr/bin/install -c log2asc /usr/local/bin/log2asc
libtool: install: /usr/bin/install -c log2long /usr/local/bin/log2long
libtool: install: /usr/bin/install -c slcan_attach /usr/local/bin/slcan_attach
libtool: install: /usr/bin/install -c slcand /usr/local/bin/slcand
libtool: install: /usr/bin/install -c slcanpty /usr/local/bin/slcanpty
make[1]: Nothing to be done for 'install-data-am'.
make[1]: Leaving directory '/home/milad/Documents/can utils/can-utils-master'

yegorich, 2015-12-23:
@milad6506 the output you show says, that all binaries were installed to /usr/local/bin.

Btw. have you tried to install can-utils via *apt install can-utils*? See http://packages.ubuntu.com/source/trusty/can-utils

marckleinebudde, 2015-12-23:
The output looks ok to me.

milad6506, 2015-12-23:
but related command are not working
like candump or ifconfig and etc

marckleinebudde, 2015-12-23:
Can you give us the error messages?

milad6506, 2015-12-24:
ok,
$ candump vcan0
SIOCGIFINDEX: No such device
$ sudo modprobe can and $ sudo modprobe vcan
i get nothing
$ sudo ifconfig can0 up
can0: ERROR while getting interface flags: No such device
please note that i want to create and send virtual CAN frames to applications on my ubuntu PC.

special thanks

yegorich, 2015-12-24:
@milad6506 can-utils are working as expected. You're just missing *vcan0* interface. Please take a look at this tutorial: http://elinux.org/Bringing_CAN_interface_up

milad6506, 2015-12-28:
hi,
i tried and here is what i got:
$ modprobe vcan
$ sudo iplink add dev vcan0 type vcan
RTNETLINK answers: File exists
$ sudo ip link set up vcan0
is it ok?

yegorich, 2015-12-28:
Please provide output from *ip addr show vcan0* after your last command. Now open two terminal windows. In one window execute: *candump vcan0* in another window *cansend vcan0 123#1122*. You should see candump receiving this CAN telegram.

hartkopp, 2016-01-16:
$ sudo iplink add dev vcan0 type vcan
RTNETLINK answers: File exists

1. There's a space missing: Must be 'ip link' instead if 'iplink'
2. When 'RTNETLINK answers: File exists' the 'vcan0' interface already has been created.

'sudo ip link add dev vcan123 type vcan' should work. Btw. the 'dev name' option can be left out to create just another vcan instance which is enumerated on tis own then.

Anyway there was no new reply from milad6506 for three weeks - so I assume he made it working and therefore will close this request.

closed 2016-01-16 by hartkopp

mohammed-barghash, 2016-05-19:
when I try to send or receive through isotp it gives me this error 
can: request_module (can-proto-6) failed.
socket: Protocol not supported. 

Any ideas ?
Thanks in advance

hartkopp, 2016-05-19:
Please follow this documentation:
https://github.com/hartkopp/can-isotp-modules/blob/master/README.isotp

The can-isotp module does not load automatically if not patched into the kernel.

mohammed-barghash, 2016-05-19:
Thank you for your reply. 

I followed the documentation and tried to load the ISO-TP module into the kernel but it gives me this error:

can_isotp: disagrees about version of symbol module_layout
insmod: ERROR: could not insert module ./can-isotp.ko: Invalid module format

any suggested solution ?

Thanks in advance

hartkopp, 2016-05-19:
You did not compile against the running kernel - or you use a different compiler.

https://www.google.de/search?q=could+not+insert+module+Invalid+module+format&ie=utf-8&oe=utf-8

finds e.g.
http://stackoverflow.com/questions/17560989/insmod-failing-to-insert-a-really-simple-module