linux-can/can-utils #13: canfdtest utility bug

rshanmu, 2016-01-08:
canfdtest utility is using gettimeofday call to inject a 3msec delay in a tight loop. This is buggy for many reasons

- gettimeofday usage on systems running ntpd can cause jumps 
- the tight loop logic also needs fixing. Why not use millisleep? 

                                gettimeofday(&tvn, NULL);
                                while ((tv_stop.tv_sec > tvn.tv_sec) ||
                                       ((tv_stop.tv_sec = tvn.tv_sec) &&
                                        (tv_stop.tv_usec >= tvn.tv_usec)))
                                        gettimeofday(&tvn, NULL);

For e.g. if tv_stop.tv_usec is 999999 and tv_usec is also 999999, the code enters the loop again and calls gettimeofday. The only way we can come of the loop is when tv_usec is 0 or more, which means the tv_sec is incremented => infinite loop. This is what happening in my system

marckleinebudde, 2016-01-08:
Yes, `gettimeofday()` is problematic here, when it comes to ntp or someone else to set the clock. We can easily convert this to `usleep()`. This has the drawback that it's interrupted by a signal. The better approach would be `clock_gettime()` and `clock_nanosleep` with `CLOCK_MONOTONIC` and an absolute (`TIMER_ABSTIME`) timeout.
What do you think?

rshanmu, 2016-01-08:
Yes, I agree. `clock_nanosleep` with `CLOCK_MONOTONIC` would be the better choice

hartkopp, 2016-01-16:
Would you like to provide a patch which moves to code to use clock_nanosleep ?

rshanmu, 2016-03-21:
I have made a pull request for this issue.

Apologies for the overwrites. Missed formatting on the last one.

closed 2016-11-21 by hartkopp