linux-can/can-utils #18: slcand arbitrary serial-can controller

reinzor, 2016-04-22:
Hi all,

Up until now, I've been working with SocketCAN via CANUSB adapters from Lawycel (http://www.can232.com/?page_id=16). However, now we received a new platform with another serial-can controller from axiomtek (http://www.axiomtek.com/Default.aspx?MenuId=Products&FunctionId=ProductView&ItemId=8270).

I am able to communicate with the CAN controller via serial and receive CAN frames. However, using slcand to create a socketcan interface does not work. As I see here https://github.com/linux-can/can-utils/blob/master/slcand.c , the slcand program sends specific commands to the serial bus and then creates a bridge between the socket and serial. However, the commands are different for my CAN controller.

Can somebody give me some pointers how to get SocketCAN to work with an arbitrary serial-can controller? 

I'm am very new to these low level Linux things so I really would appreciate your help / corrections. Thanks in advance,

-Rein

reinzor, 2016-04-27:
Anwered via mailing list

closed 2016-04-27 by reinzor

Mike351c, 2016-04-30:
To connect via can bus. The proper serial kick code must be obtained either
data acquisition or serial data recorder.
Industry has made the can bus propriety for code reading, but each
manufacturer uses a different kick to get deeper into the bus.

Mike

arasbm, 2016-09-06:
@reinzor could you share the response you got on the mailing list? I am running into a very similar issue. I am working with a PC with built-in ODB-II port and it shows up as `tty03`. However, I can not find any way to map this serial port using socketcan and canutils. Can you share some details about how you solved your problem?

reinzor, 2016-09-07:
@arasbm ; this is the mail response:

> Hello Rein,
> 
> > Up until now, I've been working with SocketCAN via CANUSB adapters
> > from Lawycel (http://www.can232.com/?page_id=16). However, now we
> > received a new platform with another serial-can controller from
> > axiomtek (http://www.axiomtek.com/Default.aspx?MenuId=Products&FunctionId=ProductView&ItemId=8270).
> > 
> > I am able to communicate with the CAN controller via serial and
> > receive CAN frames. However, using slcand to create a socketcan
> > interface does not work. As I see here
> > https://github.com/linux-can/can-utils/blob/master/slcand.c , the
> > slcand program sends specific commands to the serial bus and then
> > creates a bridge between the socket and serial. However, the commands
> > are different for my CAN controller.
> 
> Yes. That's pretty bad to reinvent the wheel by creating 'just another' CAN ASCII protocol :-(
> 
> > Can somebody give me some pointers how to get SocketCAN to work with
> > an arbitrary serial-can controller?
> 
> The slcan driver supports the standard SLCAN protocols as you know from Lawicel:
> 
> https://github.com/linux-can/can-misc/blob/master/docs/SLCAN-API.pdf
> 
> > I'm am very new to these low level Linux things so I really would
> > appreciate your help / corrections. Thanks in advance,
> 
> You would need to extend the slcan.c driver and the slcan tools from the can-utils to cope with the Axiomtek protocol.
> 
> Alternatively you can write a user space application that talks to the Axiomtek device and send the CAN frames to a virtual CAN (e.g. vcan0) so that you can use the can-utils on top of that virtual CAN.
> 
> Just a different idea: As you obviously have a full-size mini PCI-Express module - why don't you use a proper CAN PCIe interface with a PCI-connected SJA1000 controller which is already supported by Linux?
> 
> All this serial line encapsulation stuff does not support the really good (configuration) things and error handling from SocketCAN.
> 
> Regards,
> Oliver
> 
> ps. Maybe Axiomtek can also provide a SLCAN compatible firmware ...

At the end, we did not continue with this specific controller since it was not compatible with the SLCAN protocol specified here https://github.com/linux-can/can-misc/blob/master/docs/SLCAN-API.pdf