linux-can/can-utils #22: lib.h -- please utilize pointer to const

ajneu, 2016-06-13:
Hi,

thanks for your nice library!

Please could the following function prototypes be changed, to have to bold **`const`**:

* `void snprintf_can_error_frame(char *buf, size_t len, `**`const `**`struct canfd_frame *cf, `**`const `**`char* sep)` ([ref](https://github.com/linux-can/can-utils/blob/master/lib.c#L566))   
  
* `static int snprintf_error_lostarb(char *buf, size_t len, `**`const `**`struct canfd_frame *cf)` ([ref](https://github.com/linux-can/can-utils/blob/master/lib.c#L522))   

* `static int snprintf_error_ctrl(char *buf, size_t len, `**`const `**`struct canfd_frame *cf)` ([ref](https://github.com/linux-can/can-utils/blob/master/lib.c#L529))   

* `static int snprintf_error_prot(char *buf, size_t len, `**`const `**`struct canfd_frame *cf)` ([ref](https://github.com/linux-can/can-utils/blob/master/lib.c#L545))

Reason: I'm using compiler warnings, and my compile is tripping up.

Here's a patch (should work if you do: `git apply patchfile`)

**patchfile**
```patch
diff --git a/lib.c b/lib.c
index 973be52..fe18e0f 100644
--- a/lib.c
+++ b/lib.c
@@ -519,14 +519,14 @@ static int snprintf_error_data(char *buf, size_t len, uint8_t err,
 	return n;
 }
 
-static int snprintf_error_lostarb(char *buf, size_t len, struct canfd_frame *cf)
+static int snprintf_error_lostarb(char *buf, size_t len, const struct canfd_frame *cf)
 {
 	if (len <= 0)
 		return 0;
 	return snprintf(buf, len, "{at bit %d}", cf->data[0]);
 }
 
-static int snprintf_error_ctrl(char *buf, size_t len, struct canfd_frame *cf)
+static int snprintf_error_ctrl(char *buf, size_t len, const struct canfd_frame *cf)
 {
 	int n = 0;
 
@@ -542,7 +542,7 @@ static int snprintf_error_ctrl(char *buf, size_t len, struct canfd_frame *cf)
 	return n;
 }
 
-static int snprintf_error_prot(char *buf, size_t len, struct canfd_frame *cf)
+static int snprintf_error_prot(char *buf, size_t len, const struct canfd_frame *cf)
 {
 	int n = 0;
 
@@ -563,8 +563,8 @@ static int snprintf_error_prot(char *buf, size_t len, struct canfd_frame *cf)
 	return n;
 }
 
-void snprintf_can_error_frame(char *buf, size_t len, struct canfd_frame *cf,
-			      char* sep)
+void snprintf_can_error_frame(char *buf, size_t len, const struct canfd_frame *cf,
+                  const char* sep)
 {
 	canid_t class, mask;
 	int i, n = 0, classes = 0;
diff --git a/lib.h b/lib.h
index 4f73743..36c3a09 100644
--- a/lib.h
+++ b/lib.h
@@ -203,8 +203,8 @@ void sprint_long_canframe(char *buf , struct canfd_frame *cf, int view, int maxd
  *
  */
 
-void snprintf_can_error_frame(char *buf, size_t len, struct canfd_frame *cf,
-			      char *sep);
+void snprintf_can_error_frame(char *buf, size_t len, const struct canfd_frame *cf,
+                  const char *sep);
 /*
  * Creates a CAN error frame output in user readable format.
  */

```

Thanks.

marckleinebudde, 2016-06-13:
Can you please send a pull request?

marckleinebudde, 2016-06-13:
merged with 862e8ed1780014069797c3c2bb7d13594246822a

closed 2016-06-13 by marckleinebudde