linux-can/can-utils #26: candump exits when device goes down

ajneu, 2016-08-30:
Hi there,

`candump` is a useful tool, but the fact that
```bash
candump vcan0
```
will just exit and print `read: Network is down`, when I pull the device down (via `sudo ip link set vcan0 down`) is unfortunate.

PS: did you know, that I don't get the message `read: Network is down` and exit, when using `any` as device?? ;) Unfortunately using `any`... does not report on down/up status.

Minor aside:
There's a common use-case for binaries using [libsocketcan](http://git.pengutronix.de/?p=tools/libsocketcan.git;a=summary), to pull the device down (`can_do_stop`) and back up again (`can_do_start`). It's the only way I know of, to clear the socketcan transmit buffer. (And I really want to clear pending messages from it, because if a CAN-bus reconnect only occurs in say 5 minutes, then I don't want to transmit 5-minute old frames on reconnection.)

To try to work around the binary just exiting, I scripted up something, where I use `ip monitor` to detect when a device goes down and back up, and then invoke `candump` again. Try:
```bash
ip monitor link | while IFS=':' read num name status; do
    num=$(echo    $num    | xargs) # strip whitespace
    name=$(echo   $name   | xargs) # strip whitespace
    status=$(echo $status | xargs) # strip whitespace
    echo "   num   ==$num"
    echo "   name  ==$name"
    echo "   status==$status"
    read extraline
    ## ...
    ## ... code to handle change of status from DOWN to UNKOWN
    ## ...
done
```
But it's an ugly solution, because that script is so slow that the immediate frames sent when the device is back up *are just missed*. That's not ok.

So I decided to modify the `candump.c` code.
I've got 2 versions:
* https://github.com/ajneu/can-utils/tree/no_exit_on_down -- extremely simple
* https://github.com/ajneu/can-utils/tree/down_up_reporting -- quite nice, since I report when a device goes up or down

I believe it would be nice for something like the bottom one to go into upstream here.
I'm using some code (function `send_dump_request`) from [libsocketcan.c](http://git.pengutronix.de/?p=tools/libsocketcan.git;a=blob;f=src/libsocketcan.c;hb=HEAD). I think it's compatible with the very permissive Volkswagen license of `candump.c`.

Please pull the bottom one ([down_up_reporting](https://github.com/ajneu/can-utils/tree/down_up_reporting)) and give it a try. Pull down devices from a second shell and bring them up again. What do you think?

marckleinebudde, 2016-08-31:
Please rebase down_up_reporting on current master.

marckleinebudde, 2016-08-31:
I've addes some remarks to the patch in down_up_reporting

ajneu, 2016-08-31:
Thanks for your comments...
I've added your suggestions (C99 struct initializers etc). Please see the pull-request (29).

closed 2016-09-07 by hartkopp