linux-can/can-utils #27: Compiling can-utils on Mac OSX El Capitan.

alan707, 2016-08-31:
I've gotten quite far trying to compile can-utils on Mac OSX. I had to redefine `typedefs` and the location of `socket.h` for Mac OSX.  I finally got stuck with the following error: 

    bcmserver.c:234:8: error: no member named 'ifr_ifindex' in 'struct ifreq'
                        ifr.ifr_ifindex = caddr.can_ifindex;
                        ~~~ ^
    bcmserver.c:235:14: error: use of undeclared identifier 'SIOCGIFNAME'
                        ioctl(sc, SIOCGIFNAME, &ifr);
                                  ^
    bcmserver.c:344:19: error: use of undeclared identifier 'SIOCGIFINDEX'
                        if (!ioctl(sc, SIOCGIFINDEX, &ifr)) {
                                       ^
    bcmserver.c:345:29: error: no member named 'ifr_ifindex' in 'struct ifreq'
                                caddr.can_ifindex = ifr.ifr_ifindex;

Any help would be greatly appreciated!  I'll post my changes on my repo and would be really interested in your feedback.

marckleinebudde, 2016-08-31:
These are Linux ioctl()s to map from an interface name to the index. As far as I know socketcan is only available on Linux. Why do you want to build it on macOS?

alan707, 2016-08-31:
I wanted to try if it would compile on my laptop rather than having to use a Raspberry Pi as an intermediary.  "Since OS X is POSIX compliant, many software packages written for the other Unix-like systems such as Linux can be recompiled to run on it, including much scientific and technical software." [src](https://en.wikipedia.org/wiki/OS_X)

Is there any way you think I could make it work?

marckleinebudde, 2016-08-31:
Where do you want to run the can-utils?
You cannot use them on macOS, if you want to run them on the your rpi, you have to install a macOS -> Linux cross compilation toolchain.

alan707, 2016-08-31:
Ideally it would be great if I could run this directly from my Mac OS.  Couldn't I install can-utils on the raspberry pi and ssh into it?

marckleinebudde, 2016-08-31:
Does your macOS have CAN?

alan707, 2016-08-31:
I would connect to a device that has CAN via serial usb using slcan

marckleinebudde, 2016-08-31:
The kernel on the macOS doesn't support socketcan. You have to use a linux machine or port socket-can to the macOS kernel or write a slcan userspace driver.

closed 2016-09-05 by marckleinebudde