linux-can/can-utils #30: cansniffer filtering

vsboost, 2016-09-04:
Probably a dumb question but when running cansniffer on gmlan 33kbps its not displaying IDs that are larger than 3.

Example when running candump i get the below results, but when i run cansniffer it only displays the ones with three in this case 624#  is there anyway for it to work with larger Id's ?

(1472971756.787337) can0 10210040#0000155900001582
(1472971756.797509) can0 10220040#0000000000000000
(1472971756.807138) can0 1022E040#A14665F561F050E6
(1472971756.817105) can0 10230040#232D4AC772C265D8
(1472971756.834293) can0 624#0020000000000000
(1472971756.837776) can0 1020C040#0600002A01
(1472971756.842273) can0 10B06060#00041E0700000223
(1472971756.846630) can0 10240040#83F67701003A0FE8
(1472971756.851280) can0 10400060#820801AAD4
(1472971756.854677) can0 102C0040#629E8C2A03
(1472971756.858994) can0 10ACE060#1EC0000000000000
(1472971756.863225) can0 102CA040#000008970800F101

hartkopp, 2016-09-04:
The cansniffer makes use of CAN_BCM sockets to offload the data filtering effort into the Linux kernel. For that reason the cansniffer opens a singe CAN_BCM socket instance and creates filters for all 2048 11 bit CAN IDs. That's the reason why the current cansniffer only deals with 11 bit identifiers.
On gmlan and also on FCA vehicles most of the traffic uses 29 bit CAN IDs. That's why I started a rework of cansniffer to support 29 bit IDs (which uses the CAN_RAW socket then).
But the stuff in https://github.com/hartkopp/can-utils did not get very far until now :-(

Maybe canqv from Kurt can help you https://github.com/kurt-vd/canqv

Or you can define some routing from can0 to vcan0 with cangw and modify the CAN IDs from 29 bit ones to 11 bit ones then. But then you would not see the original CAN IDs on cansniffer from vcan0.

Looks like I have to pick up my cansniffer rework activities again :-)

vsboost, 2016-09-04:
Thanks hartkopp, ill definitely keep an eye on your work, and thanks for the explanation.

closed 2016-09-05 by vsboost

mortedamos, 2017-07-10:
Any progress on support for 29 bit CAN arbitration ids?

Mike351c, 2017-07-24:
It has been corrected in the past. This was an issue for more than one.
Hardware I'm thinking.
Git hub shell might show more.

hartkopp, 2017-07-24:
Hm - the question from https://github.com/mortedamos is valid.
The current cansniffer uses the CAN broadcast manager (CAN_BCM) inside the Kernel. For that reason (2^11) 2048 receive filters are programmed to the BCM socket. Of course you won't create 2^29 (536870912) BCM RX filters in the Kernel. Therefore the cansniffer would need a rework to work on CAN_RAW sockets and do all the filtering in userspace :-(
I started the rework in a fork: https://github.com/hartkopp/can-utils 
I think I need to push myself to start over ...  I got some other requests for this improvement recently ;-)

Mike351c, 2017-07-24:
Good input it may not be a 29 bit packet as well

coryjfowler, 2018-07-29:
I was successful in getting extended IDs working last night and I have published it with most of the basic functionality that CAN_BCM gave.  (IDs are sorted numerically, and the rx_change timeout is working).

asbachb, 2018-08-27:
I'm a little bit unsure about this issue. Is there any workaround for it? I already wondered that `candump` showed a lot of traffic but `cansniffer` showed no traffic at all after some time.
Isn't it worth to reopen this issue? When I get it right it still persists.

coryjfowler, 2018-08-28:
This issue is closed, but to elaborate on your issue...

> `candump` showed a lot of traffic but `cansniffer` showed no traffic 

This sounds like you are on a bus that uses only extended IDs.  The current version of `cansniffer` contained in this repository only supports 11 bit "Standard" IDs.  I have a fork of `can-utils` that has a version of  `cansniffer` that supports 29 bit "Extended" IDs.  Eventually this work will get merged here and thus become mainstream.

asbachb, 2018-08-28:
Just for reference: https://github.com/linux-can/can-utils/pull/88

marckleinebudde, 2024-12-10:
How to reproduce? `flexcan0` connected to `mcp251xfd0`.

```
$ cansend flexcan0 0F1FD001#1122334455667788
$ cansend flexcan0 0F1FD000#1122334455667788
```

gives:

```
$ candump any,0:0,#FFFFFFFF -cexdtA -H
 (1970-01-01 01:00:00.000000)    flexcan0  TX - -  0F1FD001   [8]  11 22 33 44 55 66 77 88
 (2024-12-10 12:15:40.879174)  mcp251xfd0  RX - -  0F1FD001   [8]  11 22 33 44 55 66 77 88
 (1970-01-01 01:00:00.000000)    flexcan0  TX - -  0F1FD000   [8]  11 22 33 44 55 66 77 88
 (2024-12-10 12:15:43.882827)  mcp251xfd0  RX - -  0F1FD000   [8]  11 22 33 44 55 66 77 88
```

```
$ cansniffer mcp251xfd0
11|ms | -- ID -- | data ...     < mcp251xfd0 # l=20 h=100 t=500 slots=1 >
99999 | 0F1FD000 | 11 22 33 44 55 66 77 88 ."3DUfw.
```

robin48gx, 2024-12-10:
> I have noticed with the CANABLE V2.0 and the kernel module for raspberry pi for MCP2518 that the 29 bit IDs where the message number is large (in my case 0x0F1FD000 or 0x0F1FD001) simply do not get shown in cansniffer or candump. I hace PIC18, STM32G431 and MCP2515 CAN modules here and they all see these higher messages. Is there a way to order filtering OFF in the kernel modules or the can0 interface ?

Sorry this may be a SJW issue. I do wonder what SJW is set to in the MCP251xfd set up. After years of having  canbus problems due to various ceramic resonators, we tend to set the SJW to maximum. this makes it more tolerant of timing sources being up to 1% different!

marckleinebudde, 2024-12-10:
Check yourself with `ip --details link show can0`. Starting with Linux v6.3 SJW is set to TSEG2 / 2, See [80bcf5ec9927](https://github.com/torvalds/linux/commit/80bcf5ec9927) ("can: bittiming: can_sjw_set_default(): use Phase Seg2 / 2 as default for SJW")

robin48gx, 2024-12-16:
Strangely for the CANABLE V2.0 device, the SJW is not shown. I can only assume changing it
from a low default value (almost certainly a low number because some of my CAN messages are going missing).
Problem solved though I think. This is an SJW issue, I just don't seem to be able to alter it with this
particular USB converter.

```
$ sudo ip --details link show can0
4: can0: <NOARP,UP,LOWER_UP> mtu 16 qdisc pfifo_fast state UP mode DEFAULT group default qlen 10
    link/can  promiscuity 0 minmtu 0 maxmtu 0 
    can state ERROR-ACTIVE restart-ms 0 
	  bitrate 100000
	     [   10000,    20000,    50000,   100000,   125000,   250000, 
	        500000,   800000,  1000000 ]
	  clock 0 numtxqueues 1 numrxqueues 1 gso_max_size 65536 gso_max_segs 65535 
 RPI4 2024-12-16_10:09 ~  
```

marckleinebudde, 2024-12-16:
Are you using the CANABLE V2.0 in `slcan` mode? That only supports fixed bit rates. So you cannot set SJW manually.

robin48gx, 2024-12-16:
> Are you using the CANABLE V2.0 in `slcan` mode? That only supports fixed bit rates. So you cannot set SJW manually.

Yes. that makes sense then. Surprised you can't set things like SJW via slcan though.

marckleinebudde, 2024-12-16:
The `slcan` API only provides a list of fixed bit rates. However you can use `slcan_attach -b` to directly set the bit timing registers, but it's not standardized how these raw values are interpreted, so they are highly depend on the used hardware.

Long story short, don't use devices with `slcan` firmware, better flash a `gs_usb` compatible firmware to your device.

robin48gx, 2025-01-08:
Just a question really. What SJW does slcan set? Or does it simply use the default value of 1?

marckleinebudde, 2025-01-08:
`slcan` doesn't set the SJW, the firmware of the USB-device does it.