linux-can/can-utils #34: bcmserver: long interface name

neozio, 2016-12-19:
I've tring to use bcmserver with long interface name over 10 character, but it didn't work well.
I found a code looks like a bug.

285                         items = sscanf(buf, "< **%6s** %c %lu %lu %x %hhu "
286                                        "%hhx %hhx %hhx %hhx %hhx %hhx "
287                                        "%hhx %hhx >",
288                                        ifr.ifr_name,
289                                        &cmd,
290                                        &msg.msg_head.ival2.tv_sec,

In the <if.h>, IFNAMSIZ is defined as 16. so I think if_name size would be 15.
Is the ifr_name size(6) is intended?

hartkopp, 2016-12-19:
You are right. The value should be 15 - or even better: It should directly depend on IFNAMSIZ-1.
I'll fix this - or would you like to provide a patch?

Thanks,
Oliver

jihochu, 2016-12-19:
I made pull request. 
please check the codes.

hartkopp, 2017-01-14:
Patch applied.

closed 2017-01-14 by hartkopp