linux-can/can-utils #40: canplayer error: No buffer space available

alan707, 2017-05-03:
I've been working quite a bit with cansniffer, cangen and candump.  I noticed that canplayer seems to have an issue when I try to replay a file.  I almost always get:
`sendto: No buffer space available`

If I type a command like
`canplayer -v <name_of_file> can0=stdout`
It correctly plays the candump file on the screen but obviously is not sending it.

My initial reaction was that the microcontroller connected to the CAN bus (a STM32F205RGY6 120Mhz ARM Cortex M3) could not take that much data in _its_ buffer size.  To dispel this, I tried sending messages at a high rate with `cangen` and was able to send messages every 3 ms which is much faster rate than recorded messages from `candump`.

Any thoughts on how I can avoid this error?

Mike351c, 2017-05-03:
There was an early bit limitation. Some late model GM will give errors.
Most are working on Euro Bosch. This pgm will open higher bit messages
though.
Let's see how it works out.
Best

alan707, 2017-05-04:
The only way I can make it work is if I split the `.log` file into 8 lines each

Mike351c, 2017-05-04:
You have found the culprit. The can message size is in compatible. I tink
last we were on 0-15 bit your message might be a 0-31 bit for some gm.
It has not been upgraded but you can de+cypher but slow things down.
Be patient or go to get hub for subsequent macro.. we should fix here first
though.
Best M

Mike351c, 2017-05-04:
I was concerned with this phenomena after reading about it over a year ago.
I have mostly Euro Vehicles Bosch fitted and it never bothered me but We
have a newer GM now but so far so good. The new BeagleBone Blue has a lot
of nice features like less peripheral requirements but the processor has
not changed much.
I believe the TT cape is capable of more but do we have the real estate on
BB Blue? Hmmm!!

Anyone?

alan707, 2017-05-12:
The problem can be solved by increasing the transmit queue length `sudo ifconfig can0 txqueuelen 1000`.  Would it make sense to explain a way to solve the buffer space problem in the error text?

closed 2018-01-19 by alan707

liviuk2, 2018-06-02:
@alan707 I agree with you to explain the possible solution it in the error text, I spent to much time to find it...