linux-can/can-utils #41: Using can-utils with OBDLink SX cable

crossband, 2017-05-30:
Hi folks!

I have realised that I can actually sniff raw CAN messages using OBDLink SX diagnostic cable and serial terminal. Could it be also done "can-utils"? I have tried so far, nothing worked. But I am not a Linux guy, maybe I did something wrong.

I like the way how "can-utils" works. Just would like to be able filter the messages.

Thanks!

hartkopp, 2017-05-31:
Hi!

You need to have a CAN network interface to use the Linux CAN infrastructure and can-utils.
The OBD CAN adapters with ELM317 chip provide some 'intelligent' diagnosis access via serial driver interface and not just 'raw CAN frames' like other Linux CAN driver adapters supported in Mainline Linux.

As you already managed to get raw CAN frames from your adapter you can send this content to a virtual CAN driver (e.g. vcan0) and use can-utils on this vcan0. An example how to do this in userspace can be seen in slcanpty.c 

This is not really a cool solution - but it's worth a try. Implementing a CAN driver analogue to the slcan.c driver in Linux would be the better solution based on the experiences from this proof of concept.

crossband, 2017-05-31:
Great idea. I'll give it a try and come back to this soon.

Thank you doing great work!

closed 2018-10-24 by hartkopp