linux-can/can-utils #42: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]

crossband, 2017-06-09:
I am trying to install can-utils on Raspberry Pi. How to solve it?

`pi@raspberrypi:~/can-utils $ make
  CC       lib.lo
  CC       canframelen.lo
  CCLD     libcan.la
  CC       asc2log.o
  CCLD     asc2log
  CC       bcmserver.o
  CCLD     bcmserver
  CC       can-calc-bit-timing.o
  CCLD     can-calc-bit-timing
  CC       canbusload.o
  CCLD     canbusload
  CC       candump.o
candump.c: In function ‘main’:
candump.c:692:21: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
       tv = *(struct timeval *)CMSG_DATA(cmsg);
                     ^
candump.c:694:7: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
       dropcnt[i] = *(__u32 *)CMSG_DATA(cmsg);
       ^
  CCLD     candump
  CC       canfdtest.o
  CCLD     canfdtest
  CC       cangen.o
cangen.c: In function ‘main’:
cangen.c:404:4: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
    *(unsigned long*)(&frame.data[0]) = random();
    ^
cangen.c:405:4: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
    *(unsigned long*)(&frame.data[4]) = random();
    ^
  CCLD     cangen
  CC       cangw.o
  CCLD     cangw
  CC       canlogserver.o
  CCLD     canlogserver
  CC       canplayer.o
  CCLD     canplayer
  CC       cansend.o
  CCLD     cansend
  CC       cansniffer.o
cansniffer.c: In function ‘rx_setup’:
cansniffer.c:373:2: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
  U64_DATA(&txmsg.frame) = (__u64) 0xFFFFFFFFFFFFFFFFULL;
  ^
cansniffer.c: In function ‘handle_keyb’:
cansniffer.c:481:4: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
    U64_DATA(&sniftab[i].notch) = (__u64) 0;
    ^
cansniffer.c: In function ‘handle_bcm’:
cansniffer.c:521:2: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
  U64_DATA(&sniftab[id].marker) |= 
  ^
cansniffer.c:522:3: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
   U64_DATA(&sniftab[id].current) ^ U64_DATA(&sniftab[id].last);
   ^
cansniffer.c:522:3: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
cansniffer.c: In function ‘handle_timeo’:
cansniffer.c:551:4: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
    U64_DATA(&sniftab[i].notch) |= U64_DATA(&sniftab[i].marker);
    ^
cansniffer.c:551:4: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
cansniffer.c:572:9: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
         U64_DATA(&sniftab[i].marker) = (__u64) 0;
         ^
cansniffer.c: In function ‘print_snifline’:
cansniffer.c:673:2: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
  U64_DATA(&sniftab[id].marker) = (__u64) 0;
  ^
  CCLD     cansniffer
  CC       isotpdump.o
  CCLD     isotpdump
  CC       isotpperf.o
  CCLD     isotpperf
  CC       isotprecv.o
  CCLD     isotprecv
  CC       isotpsend.o
  CCLD     isotpsend
  CC       isotpserver.o
  CCLD     isotpserver
  CC       isotpsniffer.o
  CCLD     isotpsniffer
  CC       isotptun.o
  CCLD     isotptun
  CC       log2asc.o
  CCLD     log2asc
  CC       log2long.o
  CCLD     log2long
  CC       slcan_attach.o
  CCLD     slcan_attach
  CC       slcand.o
  CCLD     slcand
  CC       slcanpty.o
slcanpty.c: In function ‘pty2can’:
slcanpty.c:249:2: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
  *(unsigned long long *) (&frame.data) = 0ULL; /* clear data[] */
  ^
  CCLD     slcanpty`

hartkopp, 2017-06-09:
The compiler flags usually contain '-fno-strict-aliasing' which obviously disappeared in your setup. I made a patch that fixes up the original issues in the code. But I would like to test the changes first.

hartkopp, 2017-07-01:
Fixed in https://github.com/linux-can/can-utils/commit/4c8fb05cb4d6ddcd67299008db54af423f86fd05 .

Thanks,
Oliver

closed 2017-07-01 by hartkopp