linux-can/can-utils #43: FATAL: Module vcan not found

dshvimer, 2017-06-14:
On my normal Ubuntu system. ```modprobe vcan``` works as expected. On my Chromebook the command gives me the following error:

```
modprobe: FATAL: Module vcan not found in directory /lib/modules/4.9.4-galliumos-braswell
```

hartkopp, 2017-06-14:
Are you sure the Chromebook developers enabled the CAN support in the Linux Kernel as Debian/Ubuntu does?

dshvimer, 2017-06-14:
@hartkopp Thank you for the quick response. I was tired when I wrote that and did not clarify. I have install GalliumOS on the Chromebook. It would make sense that they disabled certain features in the Kernel, maybe to gain performance. My experience with Linux is very basic, so I have this follow up question: Is there a way for me to add CAN support into the kernel?

hartkopp, 2017-06-15:
Generally these kind of questions could be better asked in the Linux-CAN Mailing list:
linux-can@vger.kernel.org

When yo have the chance to build your own Linux kernel for GalliumOS you can just enable the CAN stuff in your build and it should work. The CAN support doesn't consume performance when it's not loaded and used ;-)

For further discussions let's move to the mailing list - if you need more infos.
Thanks,
Oliver

closed 2017-06-15 by hartkopp