linux-can/can-utils #44: question about candump

CCC12, 2017-06-27:
Hello,

I have a vehicle network adapter connected to a beagle bone blue which has an on board CAN transceiver. The VNA sends out 50k CAN frames each time it's powered up with a speed of roughly 4 frames per millisecond.  Using candump I was able to capture all 50k frames. But If I use a blocking read() to get the frames from the can0 interface:
```c
    int counter = 0;
    while (1) {
        nbytes = read(s, &frame, sizeof(frame));
        counter++;
        printf("total %d\n", counter);
    }
```
the counter didn't go all the way to 50k. And after reading the source code of candump, I changed my code to:
```c
    while (1) {
        FD_ZERO(&rdfs);
        FD_SET(s, &rdfs);
        select(s + 1, &rdfs, NULL, NULL, timeout_current);
        if (FD_ISSET(s, &rdfs)) {
            nbytes = read(s, &frame, sizeof(frame));
            counter++;
            printf("total %d\n", counter);
        }
        fflush(stdout);
    }
```
the counter did show 50k at the end of every testing. I wonder why making read() unblocking will have the effect of receiving all the frames from the socket.

Thanks,
Alvin

hartkopp, 2017-06-28:
Hi Alvin,

in both cases read() is used in a blocking mode.
I would assume the fflush(stdout) makes the debug output 'predictable short' in this case.
The printf() inside the loop is generally a bad idea as it causes workload and scheduling.
IMO the fflush() just reduces the printf() problem. Without fflush() a really fat chunk of printf's is performed at some arbitrary time which may take that long, that some frames are dropped in the socket's receive queue.

Btw. you can try to extend the socket receive buffer: https://github.com/linux-can/can-utils/blob/master/candump.c#L525

Regards,
Oliver

CCC12, 2017-06-28:
Hi Oliver,

Thank you very much for getting back to me. The way I was testing candump was redirecting the output to a log file and check if there's 50000 lines in that file. I just found out that redirecting the output of my own code using read and without fflush() will resulting a log file with 50000 lines as well. So I think the overhead is in writing to the standard console output. And thanks for pointing out how to extend the socket receive buffer, it's exactly what I needed for my application.

Thanks,
Alvin

closed 2017-06-29 by hartkopp