linux-can/can-utils #46: Module vcan not found in directory /lib/modules

cchaq, 2017-07-31:
Hi,

Trying to set up a vcan using modprobe vcan but get the following error message:
modprobe: FATAL: Module vcan not found in directory /lib/modules/4.4.0-1020-aws

both "can" and "can_raw" both work but vcan keeps giving me this error.

Is there something I'm doing wrong, as I've followed the instructions but keep getting this issue.
Please let me know if you need any further info.

Thanks

hartkopp, 2017-07-31:
Hi,

your 4.4.0-1020-aws Ubuntu Kernel has no CAN interfaces configured at compile time - not even the virtual CAN driver.

See: https://packages.ubuntu.com/xenial-updates/amd64/linux-image-4.4.0-1020-aws/filelist
There's no /lib/modules/4.4.0-1020-aws/kernel/drivers/net/can/ directory :-(

Regards,
Oliver

hartkopp, 2017-07-31:
The usual way is to compile your own Kernel, which means to download the kernel sources from that 4.4.0-1020-aws Ubuntu Kernel and compile it on your own (with CONFIG_CAN_VCAN=m set in the kernel config).

I found that here for you which you just need to adapt for your kernel 4.4.0-1020-aws and the vcan:
https://askubuntu.com/questions/168279/how-do-i-build-a-single-in-tree-kernel-module

cchaq, 2017-07-31:
Thank you again, will give this a go and provide an update once complete.

hartkopp, 2017-07-31:
No problem. As this is not an issue regarding the can-utils, I'm closing this thread now.
For linux-can related questions better use the Linux CAN mailing list: linux-can@vger.kernel.org 

Thanks,
Oliver

closed 2017-07-31 by hartkopp

dhoer, 2021-03-08:
FWIW, I had same issue on Ubuntu virtualbox.  Installing modules extra `sudo apt-get install -y linux-modules-extra-$(uname -r)` will add the missing modules including vcan kernel module.

xdoestech, 2022-05-12:
Unfortunately this is a half solution because now I am getting an "unknown device type" error running "ip link add dev vcan0 type vcan" 

anyone have any idea whats up with this?

carbohydrate, 2024-01-05:
I was getting Error: Unknown device type when running ```ip link add dev vcan0 type vcan```.  Fixed by running after step 9 above.
```ip link add type vcan```

Found solution [here](https://www.kernel.org/doc/html/latest/networking/can.html#the-virtual-can-driver-vcan)