linux-can/can-utils #47: Typo?

ricardocastano, 2017-08-05:
The example and its description don't match at https://github.com/linux-can/can-utils/blob/master/candump.c#L144 . The example reads 92345678 and the description reads 12345678.

hartkopp, 2017-08-05:
No typo :-)
You can have a CAN ID 0x123 as 11bit (SFF) identifier or as 29bit (EFF) identifier!
The distinction is made in the canid_t with the CAN_EFF_FLAG:
#define CAN_EFF_FLAG 0x80000000U /* EFF/SFF is set in the MSB */
https://github.com/linux-can/can-utils/blob/master/include/linux/can.h#L54

So 0x123 -> SFF and 0x80000123 -> EFF

And therefore the can_id 92345678 together with a can_mask with set CAN_EFF_FLAG will filter for 0x12345678 as EFF CAN ID.

ricardocastano, 2017-08-05:
Thank you! That makes all the sense. I was wondering if it was a typo because the maximum valid 29bit CAN ID is 0x1FFFFFFF and 0x92345678 is outside that range.

closed 2017-08-05 by ricardocastano