linux-can/can-utils #51: PCAN USB hardware timestamps

FabianInostroza, 2017-08-15:
Hi,

Is it possible to get the hardware timestamp transformed in to system time?, I have a simple test case 
where the timestamp given are wrong: One device sends the following messages
```
1F0#XXXX030405060708
1F1#XXXX030405060708
00ABCDEF#XXXX030405060708
```
The device waits 400 us between messages and then 10 ms to repeat the sequence.

What I get on the PC with candump is
```
(1502763522.866989) can1 1F0#8E61030405060708
(1502763522.867989) can1 1F1#8F61030405060708
(1502763522.867989) can1 00ABCDEF#9061030405060708
(1502763522.877774) can1 1F0#9161030405060708
(1502763522.878752) can1 1F1#9261030405060708
(1502763522.878752) can1 00ABCDEF#9361030405060708
```
So the time deltas are wrong because the device sends multiple CAN messages in one USB packet.

I found a commit [1] by @hartkopp that changed something related to timestamps in the pcan driver, 
undoing that commit fixes the deltas but the timestamps no longer include system time.

I tried the netdev driver supplied by PEAK system and it gives the correct timestamps and deltas but it may not always be correct because it uses a non monotonic clock.

Is there a way to get both system time and deltas valid using the driver provided by linux?

[1] https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/drivers/net/can/usb/peak_usb?id=c9faaa09e2a1335678f09c70a0d0eda095564bab

pavel-kirienko, 2017-08-15:
> Is it possible to get the hardware timestamp transformed in to system time?

FWIW (not directly related to the Linux stuff) we use the Olson time synchronization algorithm for latency compensation and time base conversion with SLCAN adapters. The algorithm allows us to convert the timestamps we receive from the adapter into an arbitrary time base while compensating for the latency in the USB interface. A working Python implementation can be found here: https://github.com/UAVCAN/pyuavcan/blob/master/uavcan/driver/timestamp_estimator.py. The paper is available at https://april.eecs.umich.edu/pdfs/olson2010.pdf.

Sorry if this is considered an off-topic.

FabianInostroza, 2017-12-09:
Thanks @pavel-kirienko

closed 2017-12-09 by FabianInostroza