linux-can/can-utils #52: CAN communication terminates replaying file using canplayer

akn9017, 2017-08-27:
I am trying to send CAN data recorded using candump on the vehicle's CAN bus using canplayer. This sometimes works while sometimes terminates the CAN communication.
Example: I recorded the data for turning the indicator on and then off. After replaying it the first time using canplayer, the entire communication stopped with no error displayed. Nothing was received using candump. After restarting the system, I could send the data using canplayer. This did turn on /off the indicators. I could do it  2 times. The third time I replayed the file using canplayer, the communication stopped again (nothing received using candump) and I had to again restart the system. 
Am I missing something here? I have not terminated the CAN cable used to connect to my system as it is short (~50cm). Could this be the issue?

matthiasbock, 2017-08-30:
The missing termination can definitely be the issue. I am using a ~10cm cable and need to terminate (with 120R) on both ends, otherwise CAN won't work.

akn9017, 2017-09-08:
Hi,

I figured out that possibly my node was going in a bus-off state. Setting recovery from this has solved the problem of the disconnection in communication.
I am using the following now to enable can0 and also set recovery from bus-off:
$ sudo ip link set can0 type can bitrate 500000 restart-ms 10

However, now there is a new issue: my computer itself crashes after sending a few canplayer commands.

Any recommendations on how to solve this?

hartkopp, 2017-09-09:
You should SOLVE your CAN bus communication problem first. The fact that your controller goes into BUS_OFF shows, that either your bitrate settings (with bitrate, SJW, etc) do not fit to the other node and/or your CAN bus is not terminated correctly.

Using restart-ms does not solve your problem!! It just seems to solve it. A good choice for restart-ms is 200ms.

Can you tell more about the second node you are sending the frames too (which hardware/software bitrate settings)?

Anyway you machine should never crash. Can you send the crash message output to the Linux CAN mailing list mailto:linux-can@vger.kernel.org .

You are not describing a can-utils issue! You need some help about your general CAN problem. This GitHub issue tracking facility is not the right place for this kind of discussions.

Regards,
Oliver

closed 2018-10-09 by marckleinebudde