linux-can/can-utils #53: reserved identifier violation

elfring, 2017-10-28:
I would like to point out that an identifier like "[`__abs_choose_expr`](https://github.com/linux-can/can-utils/blob/cddb2612b1acb38c860c0993b2513560a6d500d4/can-calc-bit-timing.c#L51 "Update candidate")" [does eventually not fit](https://www.securecoding.cert.org/confluence/display/c/DCL37-C.+Do+not+declare+or+define+a+reserved+identifier "Do not use identifiers which are reserved for the compiler implementation.") to the expected naming convention of the C language standard.
Would you like to adjust your selection for unique names?

marckleinebudde, 2017-11-06:
It's directly copied from the kernel, so let's keep it this way. http://elixir.free-electrons.com/linux/latest/source/include/linux/kernel.h#L226

closed 2017-11-06 by marckleinebudde

elfring, 2017-11-23:
How do you think about to avoid that this software [depends on undefined behaviour](https://www.securecoding.cert.org/confluence/display/c/CC.+Undefined+Behavior#CC.UndefinedBehavior-ub_106 "Avoiding undefined behaviour?")?

marckleinebudde, 2017-11-23:
Get your enhancement into the kernel first, I'll update the code then.

elfring, 2017-11-23:
* How much do you distinguish between adjustments for implementations of user processes and a popular operating system?
* Would you like to help any more to improve remaining questionable identifiers?

marckleinebudde, 2017-11-23:
The purpose of this program is to have a userspace implementation of the bitrate calculation code for improving and testing the algorithm. Feel free to send patches to code lines that are not inherited from the kernel.

elfring, 2017-11-23:
Did you copy any questionable code from a known source?

marckleinebudde, 2017-11-23:
See above:

> It's directly copied from the kernel, so let's keep it this way. http://elixir.free-electrons.com/linux/latest/source/include/linux/kernel.h#L226

Updated link: http://elixir.free-electrons.com/linux/latest/source/include/linux/kernel.h#L248

elfring, 2017-11-23:
Does the software situation change if you dared to extract useful source files from the implementation of Linux for an ordinary program?

marckleinebudde, 2017-11-23:
If there's implementation defined behaviour, it's probably better to stumble over it in userspace, than in the kernel. So I suggest to fix the kernel in the first place.

elfring, 2017-11-23:
>If there's implementation defined behaviour, …

Can you become concerned about undefined software behaviour?

>So I suggest to fix the kernel in the first place.

* Would you like to help more in convincing the corresponding maintainers to fix the affected source code anyhow?
* How do you think about to collect safe arguments before discussing such an issue with Linus finally?

marckleinebudde, 2017-11-23:
> Would you like to help more in convincing the corresponding maintainers to fix the affected source code anyhow?
Sorry, I'm busy otherwise.

elfring, 2017-11-23:
>Sorry, I'm busy otherwise.

I imagine that it might be easier to fix your own source files a bit later then.

marckleinebudde, 2017-11-23:
Sure, drop me a note if there are changes in the kernel regarding this issue.

elfring, 2017-11-23:
I guess that you can wait for a looong time before the use of reserved identifiers will be reconsidered in source files from operating system implementations.
How do you identify the source code which was not copied from there?

marckleinebudde, 2017-11-23:
Quite simple, I want to test/improve `can_calc_bittiming()`. So I started with this function and added all needed functions until it compiled.

elfring, 2017-11-23:
* The act of copying of useful source files from the implementation of Linux is simple. But did you consider any related consequences before I presented this special issue?
* Do questionable reserved identifiers occur (only) in copied source code?

marckleinebudde, 2017-11-23:
> The act of copying of useful source files from the implementation of Linux is simple. But did you consider any related consequences before I presented this special issue?

no
> Do questionable reserved identifiers occur (only) in copied source code?

probably

elfring, 2017-11-23:
>no

Would you like to increase your software development considerations for this issue?

>probably

The Linux source files provide functionality which can become also useful for uses outside the application context of an operating system. How are you going to influence the software situation further then?

marckleinebudde, 2017-11-23:
> The Linux source files provide functionality which can become also useful for uses outside the application context of an operating system. How are you going to influence the software situation further then?
I don't - I rather hack on new CAN framework functionalities.

elfring, 2017-11-23:
You influence the software situation then in ways which might not be completely compliant with the standard rules for a programming language like “C” at the moment.

How do you think about to reopen this issue?

marckleinebudde, 2017-11-23:
As this code is copied from the kernel, it's whole purpose is to have it running in userspace as smooth as possible with all the side effects, bugs and undefined behaviour as it's running in the kernel. So if there is a problem, please go ahead and fix it in the kernel.

elfring, 2017-11-23:
>…, it's whole purpose is to have it running in userspace as smooth as possible

I find this expectation reasonable.

>with all the side effects, bugs and undefined behaviour as it's running in the kernel.

It seems that you express a target conflict here.
* There are undesired “side effects”. I hope that we try to reduce or even fix them to some degree.
* Do you try to convert a Linux software module into an ordinary program which would be executed by a micro-kernel?

>So if there is a problem, please go ahead and fix it in the kernel.

I am also improving Linux source files for a while. But I imagine that this issue can exceed my software development resources.

I assume that you might have got better resources in the background to resolve discussed implementation details.

marckleinebudde, 2017-11-23:
> Do you try to convert a Linux software module into an ordinary program which would be executed by a micro-kernel?

No.

> I am also improving Linux source files for a while. But I imagine that this issue can exceed my software development resources.

Mine aswell

elfring, 2017-11-23:
>Mine aswell

Our work capacity will be always limited. But the combination can trigger additional improve possibilities, can't it?

marckleinebudde, 2017-11-23:
Not in my free time - I can send you a formal offer, though.

elfring, 2017-11-23:
Is the software that you currently develop here relevant for any system certifications?

marckleinebudde, 2017-11-23:
See above:
> The purpose of this program is to have a userspace implementation of the bitrate calculation code for improving and testing the algorithm. 

...and bringing back improved versions to the kernell

elfring, 2017-11-23:
* Which components are missing so that this software could also be improved within the application context of Linux directly?
* Do you know any contributors which care also a bit more about standard compliance?

marckleinebudde, 2017-11-23:
> Which components are missing so that this software could also be improved within the application context of Linux directly?

The algorithm can be faster compiled and started in userspace than to compile a kernel module. In userspace we can call the algorithm with the parameters of all supported drivers without having the actual hardware. The program simply loops over all supported HW and clock combinations and prints the results.

elfring, 2017-11-23:
>The algorithm can be faster compiled and started in userspace than to compile a kernel module.

This view sounds for me partly as if working with a micro-kernel could help.

>In userspace we can call the algorithm with the parameters of all supported drivers without having the actual hardware.

Can you work also with simulated “hardware” for this use case?

There are some CAN tools provided. Can the importance of corresponding standard compliance increase over time?

marckleinebudde, 2017-11-24:
> This view sounds for me partly as if working with a micro-kernel could help.

Feel free to implement this in your microkernel of choice. Personally I'm focused on Linux - and we have a working solution anyways.

> Can you work also with simulated “hardware” for this use case?

Not for this usecase, however adding CAN adapters to qemu is highly welcome.

elfring, 2017-11-24:
>… - and we have a working solution anyways.

I hope so in principle.

Would you like to increase your software development attention for aspects around standard compliance?

marckleinebudde, 2017-11-24:
Sure

elfring, 2017-11-24:
>Sure

Thanks for your general interest.

Have you got any contacts to standard organisations which can be relevant for the discussed software area?

marckleinebudde, 2017-11-24:
Not yet, if this becomes relevant during my work, I'll get in touch. Thanks.

elfring, 2017-11-24:
Do you trust any other persons more (than me) for [information about the safe usage of reserved identifiers](http://en.cppreference.com/w/c/language/identifier#Reserved_identifiers "Description for reserved identifiers")?

marckleinebudde, 2017-11-24:
I don't see how this is relevant here.

elfring, 2017-11-24:
Would you insist to recheck my claims with a look into an official standard specification document for the programming language “C”?

marckleinebudde, 2017-11-24:
As stated before - feel free to fix the code in the kernel - I want to use the same code as in the kernel.

elfring, 2017-11-24:
We are trying to improve this software to some degree in different areas for a while. But I hope that more possibilities can be considered if development efforts could be bundled or recombined.

It seems that there are also special challenges to consider if every developer is waiting on a desired action by other contributors. Can collaboration become a bit easier if the right persons will be informed so that further information by them can clarify the situation in significant ways?

marckleinebudde, 2017-11-24:
Yeah - there's no politics in open source.

elfring, 2017-11-24:
>… - there's no politics in open source.

* I disagree. - There can be software development policies involved besides the usual difficulties in (human) collaboration.
* My own “development reputation” can trigger also special reactions.
* How do you think about to ask any of “the right persons” about mentioned technical details?

marckleinebudde, 2017-11-24:
See above, I'll get in touch once any of this becomes relevant.

elfring, 2017-11-24:
Would you dare to propose the renaming of any questionable identifiers in software areas you are currently working with?

marckleinebudde, 2017-11-24:
Me? No, it's your finding and you've the background. Go for it!

elfring, 2017-11-24:
>No,

It seems that you try to push undesired work back.

>it's your finding

Only partly.

If you would use static source code analysis more, you will find various update candidates. One might wonder which of them will get another bit of software development attention.

>Go for it!

I did not find somebody so far who would clarify (or even acknowledge) my interpretation for the handling of reserved identifiers in source files from operating systems. My knowledge can be still incomplete about relevant details from the standard specification.

I hope that I can avoid false claims for this issue.

marckleinebudde, 2017-11-24:
If you are not sure about your findings it seems you have some homework to do. As I'm not the author of the code in question I cannot comment on this.

elfring, 2017-11-24:
>If you are not sure about your findings it seems you have some homework to do.

I am unsure if the distinction whether a source file belongs to an operating system (or not) can make a significant difference.

>As I'm not the author of the code in question I cannot comment on this.

Some developers can tamper with the reserved name space for reasons which could be inappropriate in a concrete situation.
I could imagine that a few of them could feel to be attacked by special changes.

It can also happen that some programming mistakes are popular for a while, can't it?

marckleinebudde, 2017-11-24:
Sure, so go and ask to them.