linux-can/can-utils #54: Fix signal handler

elfring, 2017-10-28:
The function “[exit](http://pubs.opengroup.org/onlinepubs/9699919799/functions/exit.html "Terminate a process.")” does not belong to the list of [async-signal-safe functions](https://www.securecoding.cert.org/confluence/display/c/SIG30-C.+Call+only+asynchronous-safe+functions+within+signal+handlers "Call only asynchronous-safe functions within signal handlers.").
I guess that a different program design will be needed for your function “[printstats](https://github.com/linux-can/can-utils/blob/cddb2612b1acb38c860c0993b2513560a6d500d4/canbusload.c#L119 "Update candidate")”.

marckleinebudde, 2017-11-06:
What do you suggest? Use `_exit()` instead?

elfring, 2017-11-23:
>Use `_exit()` instead?

This could be one possibility.

Would you like to consider other software design options?
- [sigwaitinfo()](http://pubs.opengroup.org/onlinepubs/9699919799/functions/sigwaitinfo.html)
- [Self-pipe trick](http://cr.yp.to/docs/selfpipe.html)
- [Signal-safe locks](https://locklessinc.com/articles/signalsafe_locks/)

marckleinebudde, 2017-11-23:
send patches

elfring, 2017-11-23:
I would prefer to clarify acceptable software design directions before contributing patches in a source code format.

marckleinebudde, 2017-11-23:
I'm not familiar with your proposed alternatives, please illustrate them in form of patches, etc...

elfring, 2017-11-23:
The proposed alternatives are already illustrated by other information sources.
Would you like to become more familiar with the choices?