linux-can/can-utils #55: Completion of error handling

elfring, 2017-10-28:
Would you like to add more error handling for return values from functions like the following?
- [fprintf](http://pubs.opengroup.org/onlinepubs/9699919799/functions/fprintf.html "Print formatted output.") ⇒ [print_usage](https://github.com/linux-can/can-utils/blob/cddb2612b1acb38c860c0993b2513560a6d500d4/canbusload.c#L91)
- [sigaction](http://pubs.opengroup.org/onlinepubs/9699919799/functions/sigaction.html "Change a signal action.") ⇒ [main](https://github.com/linux-can/can-utils/blob/cddb2612b1acb38c860c0993b2513560a6d500d4/isotpserver.c#L288)

marckleinebudde, 2017-11-06:
https://github.com/linux-can/can-utils/blob/cddb2612b1acb38c860c0993b2513560a6d500d4/canbusload.c#L91 -> no
https://github.com/linux-can/can-utils/blob/cddb2612b1acb38c860c0993b2513560a6d500d4/isotpserver.c#L288 -> yes

elfring, 2017-11-23:
I suggest to avoid [ignorance of return values](https://cwe.mitre.org/data/definitions/252.html "Checking more return values") a bit more.
Would you like to detect every error situation as early as possible?

marckleinebudde, 2017-11-23:
Feel free to send patches for the sigaction.

closed 2018-10-09 by marckleinebudde