linux-can/can-utils #56: Reconsider final actions for program termination

elfring, 2017-10-28:
I have taken another look at the end of [the implementation for the function “main()”](https://github.com/linux-can/can-utils/blob/cddb2612b1acb38c860c0993b2513560a6d500d4/log2asc.c#L69 "Update candidate: main()"). I got the impression that only functions are called which should be usually performed by [a regular program termination](http://en.cppreference.com/w/c/program/exit "Description for exit").
Thus I suggest to omit a bit of extra source code.

```diff
diff --git a/log2asc.c b/log2asc.c
index 8fb88a9..509897b 100644
--- a/log2asc.c
+++ b/log2asc.c
@@ -198,9 +198,6 @@ int main(int argc, char **argv)
                        fprintf(outfile, "\n");
                }
        }
-       fflush(outfile);
-       fclose(outfile);
-       fclose(infile);
 
        return 0;
 }
```

:crystal_ball: How do you think about to delete any unnecessary stuff also in other source files?

marckleinebudde, 2017-11-06:
I don't like this. I like to clean up after finishing work.

closed 2017-11-06 by marckleinebudde

elfring, 2017-11-23:
>I like to clean up after finishing work.

You can omit extra actions when such functionality is already provided by the programming language implementation, can't you?

marckleinebudde, 2017-11-23:
I can, but I don't have to.

elfring, 2017-11-23:
If you insist to keep three questionable function calls, I suggest that you should also [complete the corresponding error detection and exception handling for them](https://github.com/linux-can/can-utils/issues/55 "Completion of error handling").

marckleinebudde, 2017-11-23:
Feel free to send patches

elfring, 2017-11-23:
I showed the patch to delete three questionable function calls already.