linux-can/can-utils #57: Can candump log from zero?

shinhwap, 2017-11-15:
Hello,

I'm using can-utils and it works well.
The problem is I want to log the can data from zero timestamp
candump -tz can0 can show the can data from zero timestamp
But candump cannot log data from zero
I tried to type candump -tz -l can0, the timestamp still start from absolute
I saw the code and find that -tz will create a local variable: timeval diff, and log using tv.tv_sec, tv.tv_usec to fprintf the log file. 
So the -tz will not work for logging.
Is there other way to achieve logging data from zero timestamp?

hartkopp, 2017-11-15:
Hm - so far no one has ever requested such a feature. The idea was to record the logfile with the correct host timestamps to have to possibility to sync the CAN log with other logfiles in the vehicle (e.g. camera or GPS timestamps).
At replay time the canplayer starts with the first timestamp and sends the frames based on the offset from the previous timestamp.
So what would be your use-case for starting the timestamp at 0.0s ?

shinhwap, 2017-11-16:
I'm using Raspberry Pi 3 to read CAN data.
Pi doesn't have RTC and we also do not have internet connection in P3, so the timestamp in Pi will be wrong. 
I just want to record the CAN data from startup to shutdown.
It's ok if there is not a ready-made features to do that.
I will close this issue. 
Thank you for your reply!

closed 2017-11-16 by shinhwap