linux-can/can-utils #60: canplayer does not parse timestamp from log file

hauptmech, 2017-12-11:
After a lot of pain I discovered that canplayer does not parse a time in seconds from the log file. Instead it parses seconds and microseconds as integers with '.' as the seperator.

Using '.' as the separator in the timestamp is very misleading. I assumed that the timestamp was a single value in seconds. It lead to a lot of subtle errors when the timestamps I was feeding it were a floating point seconds with varying digits of precision.

Can we either change the separator so that it does not look like a single timestamp of seconds or parse the field as seconds.fractions_of_a_second?

hartkopp, 2017-12-11:
The timestamp is a representation of struct timeval - which consists of tv_sec and tv_usec:
http://pubs.opengroup.org/onlinepubs/7908799/xsh/systime.h.html
And the log file timestamp format is defined as (%010ld.%06ld):
https://github.com/linux-can/can-utils/blob/master/candump.c#L763
So the tv_usec "factions of a second" is represented in fixed and expected in 6 columns.
From that perspective the issue "canplayer does not parse timestamp from log file" does not really hit the point ;-)
Anyway: Changing the timestamp separator would brake the established logfile format - which is fixed for more than 10 years now. This would urge millions of logfiles to be converted. 
The only thing I could image is to introduce a check whether the "fractions of a digit" are exactly 6 characters, so that it terminates on 3rd party logfiles which come with a variable digits of precision.

hauptmech, 2017-12-11:
A test for exactly 6 digits would have caught my misinterpretation of the timestamp format. Sounds like a good solution. 

Happy to submit a patch but it will be a few weeks.

hartkopp, 2017-12-12:
What about this patch:

>   /\*
>    \* ensure the fractions of seconds are 6 characters long to catch
>    \* 3rd party or handcrafted logfiles that treat the timestamp as float
>    \*/
>    if (strchr(buf, ')') - strchr(buf, '.') != 7) {
>              fprintf(stderr, "incorrect timestamp format in logfile\n");
>              return 1;
>   }

That catches e.g. (123.1234567) and (123.12345) but not (123.12345 ) <- space after '5'

But this would mean that the creator of the logfile has taken care about making the fraction 6 characters long but did not make it %06d .... is this a realistic pattern?
Does the patch above detect enough in your opinion?

hauptmech, 2017-12-13:
The parenthesis imply that things inside them are parsed differently than the rest of the line. I'd assume that whitespace is ignored inside them.  However the sscanf return should fail if there is a space between digits and ')' right? 

'Incorrect timestamp format in logfile. Microseconds part of timestamp must be 6 digits.\n' saves reading the code to understand the error.

Looks good!

hartkopp, 2017-12-13:
Yes. In fact sscanf() does the rest :-)

Applied and pushed the patch.

Tnx!

closed 2017-12-13 by hartkopp