linux-can/can-utils #61: Android.mk Errors

rubberduck203, 2018-01-12:
I had to make the following patch in order to get can-utils to build inside the AOSP source tree.

```patch
diff --git a/Android.mk b/Android.mk
index 86557ae..fee085e 100644
--- a/Android.mk
+++ b/Android.mk
@@ -3,7 +3,11 @@ LOCAL_PATH := $(call my-dir)
 PRIVATE_LOCAL_CFLAGS := -O2 -g -W -Wall                \
                        -DSO_RXQ_OVFL=40        \
                        -DPF_CAN=29             \
-                       -DAF_CAN=PF_CAN
+                       -DAF_CAN=PF_CAN \
+            -Wno-error=unused-parameter \
+            -Wno-error=sign-compare \
+            -Wno-error=pointer-arith \
+            -Wno-error=address-of-packed-member
 
 #
 # canlib
```

I'm not sure if suppressing these is the right thing to do.
Wanted to check here submitting a PR.
I'd rather see this upstreamed (or the warnings fixed) than maintain a branch just for this.

Mike351c, 2018-01-12:
Hmmm, does the patch support legacy builds.
I like the idea for on board integrated systems. Linux is low real estate
but does require user to invoke or execute change which is a driver
distraction.
Very interested.

rubberduck203, 2018-01-13:
I’ve done a bit of research by looking through all of the versions of the `gcc` docs. 

https://gcc.gnu.org/onlinedocs/

The `-Wno-error=` switch was introduced in [gcc v4.2.4](https://gcc.gnu.org/onlinedocs/gcc-4.2.4/gcc/Warning-Options.html#Warning-Options). That version was released in 2005 and the latest version is 7.2. Given that information, I personally would be comfortable allowing the switch if it was one of my projects. 

Trying to put myself in your shoes, the only thing I’m unsure of is whether I would want to accept this in lieu of fixing of the actual warnings. Unfortunately I’m not sure I’m familiar enough with the code base to give it an attempt.

Mike351c, 2018-01-14:
Many of the ARM based Android builds were on lollypop, marshmallow.
Next would be to Port the Linux code to Android studio. Drivers and
interface could be a challenge.

rubberduck203, 2018-01-14:
Port the code? I’m afraid I don’t understand. 
All of the utilities we’ve installed on Oreo have worked out of the box so far. Admittedly, we’ve mostly just used `candump` and `cansend` to establish initial communication, but no problems so far. The project just works.

yegorich, 2018-01-14:
What about fixing the code, so that it won't produce such warnings?

rubberduck203, 2018-01-15:
I'm not sure I have the expertise to fix them, but I can share what they are.
They all seem to enabled by the `-Wextra` flag.
The [Makefile in the root directory][1] isn't using this flag, so perhaps it's as simple as removing the `-W` flag.

[1]: https://github.com/linux-can/can-utils/blob/cf949a926ace19c453f29797c4f0436f3e391389/Makefile#L46

Regardless, here are the warnings that are being generated.

```
[  0% 53/7269] target  C: bcmserver <= external/can-utils/bcmserver.c
external/can-utils/bcmserver.c:132:20: warning: unused parameter 'i' [-Wunused-parameter]
void childdied(int i)
                   ^    
external/can-utils/bcmserver.c:137:14: warning: unused parameter 'argc' [-Wunused-parameter]
int main(int argc, char **argv)
             ^    
external/can-utils/bcmserver.c:137:27: warning: unused parameter 'argv' [-Wunused-parameter]
int main(int argc, char **argv)
                          ^    
3 warnings generated.
[  0% 54/7269] Check module type: out/target/product/hikey960/obj/SHARED_LIBRARIES/libm_intermediates/link_type
[  0% 55/7269] Export includes file:  -- out/target/product/hikey960/obj/EXECUTABLES/canplayer_intermediates/export_includes
[  0% 56/7269] target  C: can-calc-bit-timing <= external/can-utils/can-calc-bit-timing.c
[  0% 57/7269] target  C: canbusload <= external/can-utils/canbusload.c
external/can-utils/canbusload.c:119:18: warning: unused parameter 'signo' [-Wunused-parameter]
void sigterm(int signo)
                 ^    
external/can-utils/canbusload.c:124:21: warning: unused parameter 'signo' [-Wunused-parameter]
void printstats(int signo)
                    ^    
external/can-utils/canbusload.c:300:14: warning: comparison of integers of different signs: 'int' and 'unsigned long' [-Wsign-compare]
                if (nbytes >= IFNAMSIZ+sizeof("@1000000")+1) {
                    ~~~~~~ ^  ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
external/can-utils/canbusload.c:396:16: warning: comparison of integers of different signs: 'int' and 'unsigned long' [-Wsign-compare]
                                if (nbytes < sizeof(struct can_frame)) {
                                    ~~~~~~ ^ ~~~~~~~~~~~~~~~~~~~~~~~~
4 warnings generated.
[  0% 58/7269] Import includes file: out/target/product/hikey960/obj/EXECUTABLES/canplayer_intermediates/import_includes
[  0% 59/7269] target  C: canfdtest <= external/can-utils/canfdtest.c
external/can-utils/canfdtest.c:148:32: warning: unused parameter 'signo' [-Wunused-parameter]
static void signal_handler(int signo)
                               ^
1 warning generated.
[  0% 66/7269] target  C: libcan <= external/can-utils/lib.c
external/can-utils/lib.c:585:16: warning: comparison of integers of different signs: 'int' and 'unsigned long' [-Wsign-compare]
        for (i = 0; i < ARRAY_SIZE(error_classes); i++) {
                    ~ ^ ~~~~~~~~~~~~~~~~~~~~~~~~~
1 warning generated.
[  0% 67/7269] target  C: canlogserver <= external/can-utils/canlogserver.c
external/can-utils/canlogserver.c:140:22: warning: comparison of integers of different signs: 'int' and 'size_t' (aka 'unsigned long') [-Wsign-compare]
        if (max_devname_len < strlen(ifr.ifr_name))
            ~~~~~~~~~~~~~~~ ^ ~~~~~~~~~~~~~~~~~~~~
external/can-utils/canlogserver.c:156:20: warning: unused parameter 'i' [-Wunused-parameter]
void childdied(int i)
                   ^
external/can-utils/canlogserver.c:164:23: warning: unused parameter 'i' [-Wunused-parameter]
void shutdown_gra(int i)
                      ^
3 warnings generated.

[  0% 70/7269] target  C: cangw <= external/can-utils/cangw.c
external/can-utils/cangw.c:89:8: warning: arithmetic on a pointer to void is a GNU extension [-Wpointer-arith]
        rta = NLMSG_TAIL(n);
              ^~~~~~~~~~~~~
external/can-utils/cangw.c:76:46: note: expanded from macro 'NLMSG_TAIL'
        ((struct rtattr *)(((void *) (nmsg)) + NLMSG_ALIGN((nmsg)->nlmsg_len)))
                           ~~~~~~~~~~~~~~~~~ ^
external/can-utils/cangw.c:84:49: warning: comparison of integers of different signs: 'unsigned int' and 'int' [-Wsign-compare]
        if (NLMSG_ALIGN(n->nlmsg_len) + RTA_ALIGN(len) > maxlen) {
            ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~ ^ ~~~~~~
[... 56 lines cut ...]
             ^
external/can-utils/log2long.c:54:27: warning: unused parameter 'argv' [-Wunused-parameter]
int main(int argc, char **argv)
                          ^
2 warnings generated.

[  1% 118/7269] target  C: cansniffer <= external/can-utils/cansniffer.c
external/can-utils/cansniffer.c:178:18: warning: unused parameter 'signo' [-Wunused-parameter]
void sigterm(int signo)
                 ^
external/can-utils/cansniffer.c:400:23: warning: comparison of integers of different signs: 'ssize_t' (aka 'long') and 'size_t' (aka 'unsigned long') [-Wsign-compare]
        if (read(0, cmd, 19) > strlen("+123456\n"))
            ~~~~~~~~~~~~~~~~ ^ ~~~~~~~~~~~~~~~~~~~
external/can-utils/cansniffer.c:534:22: warning: unused parameter 'fd' [-Wunused-parameter]
int handle_timeo(int fd, long currcms){
                     ^
3 warnings generated.

[  1% 125/7269] target  C: isotptun <= external/can-utils/isotptun.c
external/can-utils/isotptun.c:127:18: warning: unused parameter 'signo' [-Wunused-parameter]
void sigterm(int signo)
                 ^
1 warning generated.

```

yegorich, 2018-01-16:
@rubberduck203 OK. I see. In this case I would suggest that you create a pull request with your patch.

@hartkopp @marckleinebudde what do you think about using Travis-CI? We could also gcc, cland, arm-gcc and Android NDK build tests there and have the most common build environments covered.

yegorich, 2018-01-17:
`android-ndk-r10e` doesn't understand `-Wno-error=address-of-packed-member`, the newest version (`android-ndk-r16b`) has no problem.

I have no idea what `gcc` versions our users use.

marckleinebudde, 2018-01-17:
If you know the gcc Version, you can do something like this:

```
CXXVER := $(shell $(CXX) -dumpversion)
ifeq ($(shell expr match $(CXX) 'clang'),5)
	NEEDED_CXXFLAGS += -std=c++11
else ifeq ($(shell expr match ${CXXVER} "4\.[0-9][0-9]"),4) # gcc >= 4.10
	NEEDED_CXXFLAGS += -std=c++11
else ifeq ($(shell expr match ${CXXVER} "4\.[7-9]"),3) # >= 4.7
	NEEDED_CXXFLAGS += -std=c++11 -D_GLIBCXX_USE_NANOSLEEP=1
else ifeq ($(shell expr match ${CXXVER} "4\.6"),3) # = 4.6
	NEEDED_CXXFLAGS += -std=c++0x
else ifeq ($(shell expr match ${CXXVER} "[5-7]\.[0-9]"),3) # gcc >= 5.0
	NEEDED_CXXFLAGS += -std=c++11
else ifeq ($(shell expr match ${CXXVER} "7"),1) # gcc 7 ubuntu
	NEEDED_CXXFLAGS += -std=c++11
else # not supported
$(error Compiler too old)
endif
```

rubberduck203, 2018-01-17:
What about removing the `-Wextra` flag? The Linux Makefile doesn’t use it. Seems simpler to just be consistent.

marckleinebudde, 2018-01-17:
Where does it come from? The android buildsystem?

rubberduck203, 2018-01-17:
It's defined in the Android.mk file.

The `-W` flag.

```makefile
PRIVATE_LOCAL_CFLAGS := -O2 -g -W -Wall		\
```

https://github.com/linux-can/can-utils/blob/master/Android.mk#L3

marckleinebudde, 2018-01-17:
Ok, I was looking for `-Wextra`. The original Makefile was contributed by @yegorich. Let's wait for his ok.

yegorich, 2018-01-17:
@rubberduck203 @marckleinebudde I'm OK with removing this flag.

rubberduck203, 2018-01-17:
Cool. I’ll get the PR updated when I get a few minutes. 

Thanks for hashing this out with me!

rubberduck203, 2018-01-18:
Okay, so I'm going to abandon #63.
The AOSP build system has a [whitelist of projects that can have warnings](https://android.googlesource.com/platform/build/soong/+/master/cc/config/global.go#123).
If a project is not in that whitelist, the AOSP build system will add `-W -Wall` to the `CFLAGS`.

The real answer here is to place the project into the `device` directory.
For posterity, here is a local_manifest file that will place can-utils into a directory of the AOSP tree where warnings are allowed.

```xml
<?xml version="1.0" encoding="UTF-8"?>
<manifest>
  <remote fetch="https://github.com/" name="github" />
  <project path="device/external/can-utils" name="linux-can/canutils" remote="github" revision="master" />
```

closed 2018-01-18 by rubberduck203