linux-can/can-utils #67: Using BCM for sending/receiving canfd_frames

xatavian, 2018-01-26:
Good morning,

I am using SockatCAN on Debian and i'd like to use the BCM to send and receive frames. I successfully set up the BCM for can_frame but I'm having issues on how to setup it for canfd_frame.

The Linux kernel doc on [SocketCAN](https://www.kernel.org/doc/Documentation/networking/can.txt) gives the following information:

> 4.2.7 Broadcast Manager CAN FD support
> 
>   The programming API of the CAN_BCM depends on struct can_frame which is
>   given as array directly behind the bcm_msg_head structure. To follow this
>   schema for the CAN FD frames a new flag 'CAN_FD_FRAME' in the bcm_msg_head
>   flags indicates that the concatenated CAN frame structures behind the
>   bcm_msg_head are defined as struct canfd_frame.
> 
>     struct {
>             struct bcm_msg_head msg_head;
>             struct canfd_frame frame[5];
>     } msg;
> 
>     msg.msg_head.opcode  = RX_SETUP;
>     msg.msg_head.can_id  = 0x42;
>     msg.msg_head.flags   = CAN_FD_FRAME;
>     msg.msg_head.nframes = 5;
>     (..)

Unfortunately, I can't find any CAN_FD_FRAME flag and I could not find information for TX. Could you help me with that ?

Best regards,

hartkopp, 2018-01-26:
Hi,

please take a look at the recent commits in the can-tests repository (https://github.com/linux-can/can-tests).
E.g. commit https://github.com/linux-can/can-tests/commit/12271d45b350bb8325a08cf5e3ec0f5b2189784e introduces two BCM tools to test the functionality with CAN FD.

The CAN FD support for the BCM has been introduced in Linux 4.8:
https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux-stable.git/log/net/can/bcm.c?h=linux-4.8.y

So if you miss the CAN_FD_FRAME in bcm.h either your Linux kernel or the header files of your distribution may be out of date.

Best regards,
Oliver

xatavian, 2018-01-30:
Thank you, it actually turned out that my kernel version is out of date for this.

Best regards,

closed 2018-01-30 by hartkopp