linux-can/can-utils #68: Question: How can I automatically bring up CAN interface using netplan?

arasbm, 2018-01-30:
I apologize first as this is most likely the wrong place to ask my question but I can not find how to get in touch with socket CAN community. 

I am wondering how I could use `netplan` to bring up and configure a CAN interface. Previously I was able to automate this using the '/etc/network/interfaces' file. As of ubuntu 17.10 this file is no longer used and I think I am supposed to use `netplan` now. However, I do not know how to detect CAN interface becoming available. Have you by any chance ran into this issue? I think soon a lot of people will hit this problem, especially after 18.04 LTS is released.

I appreciate any help!
https://askubuntu.com/questions/991623/how-to-automatically-bring-up-can-interface-in-ubuntu-17-10

This was my script that used to work:

```
$ cat /etc/network/interfaces
# interfaces(5) file used by ifup(8) and ifdown(8)
auto lo
iface lo inet loopback

allow-hotplug can0
iface can0 can static
    bitrate 250000
        up /sbin/ip link set $IFACE down
    up /sbin/ifconfig $IFACE txqueuelen 20000
    up /sbin/ip link set $IFACE type can bitrate 250000 restart-ms 10
        up /sbin/ip link set $IFACE up
```

marckleinebudde, 2018-01-30:
As far as I know `netplan` is some kind of a config file generator for different backends like `NetworkManager` or `sytsemd-networkd`. `systemd-networkd` doesn't have support for CAN yet, although there's some work in progress code for [`systemd`](https://github.com/marckleinebudde/systemd/tree/can). I don't if there's CAN support for NM.
So you first need to add CAN support to the backend you're using. If it's NM, you can probably call scripts to configure CAN as you do now.

hartkopp, 2018-01-30:
For the old Debian/Ubuntu init system, you might also use the can_if script
https://github.com/linux-can/can-misc/blob/master/etc/can_if

Just copy can_if to /etc/init.d/can_if

and invoke

update-rc.d can_if defaults

The configuration options are documented inside the can_if script.

Stefan-Steinl, 2019-06-04:
Are there any new insights in the topic of specifically using netplan (no workarounds) with a socket_can interface? I wanted to set up something similar.

marckleinebudde, 2019-06-04:
Meanwhile, you can use `systemd-networkd` to configure CAN interfaces automatically.

AndySchroder, 2020-02-11:
> For the old Debian/Ubuntu init system, you might also use the can_if script
> https://github.com/linux-can/can-misc/blob/master/etc/can_if
> 
> Just copy can_if to /etc/init.d/can_if
> 
> and invoke
> 
> update-rc.d can_if defaults
> 
> The configuration options are documented inside the can_if script.
> 
> ps. I have a Debian testing here with systemd - but the old scripts still work as expected.

Does this work only on boot or does it also work if you USB hot plug a can interface?

AndySchroder, 2020-02-11:
> Meanwhile, you can use `systemd-networkd` to configure CAN interfaces automatically.

Does this work on ubuntu 18.04, and are you willing to share an example? Does this work only on boot or does it also work if you USB hot plug a can interface?

marckleinebudde, 2020-02-11:
Place this file for example in `/etc/systemd/network/80-can.network`

```
[Match]
Name=can*

[CAN]
BitRate=500K
```

It works with both hot and cold plug devices.

closed 2020-02-20 by arasbm

rfrowe, 2020-03-03:
> ```
> BitRate=500K
> ```

This is invalid syntax. I had to use `500000` else `systemd-networkd.service` fails to start.

marckleinebudde, 2020-03-03:
I'm using this file on my laptop and it works

```
➜ (pts/7) frogger@hardanger:~ (master) ✗ cat /etc/systemd/network/80-can.network 
[Match]
Name=can*

[CAN]
BitRate=500K
```

rfrowe, 2020-03-04:
Huh weird. Whatever floats your boat, I guess. My boat was Ubuntu Server 19.04.

janblumenkamp, 2021-08-16:
Is there a way to set the TX Queue length using the network manager, similar to this command?
```
sudo ifconfig can0 txqueuelen 1000
```
It doesn't look like there is an [option for this in the network manager](https://man7.org/linux/man-pages/man5/systemd.network.5.html#[CAN]_SECTION_OPTIONS)... How else can I set this automatically during boot?

marckleinebudde, 2021-08-16:
The man page you've linked to is systemd-networkd, not Network Manager. I think you can set this automatically during boot using udev (not tested though):

```
cat <<EOF>> /etc/udev/rules.d/71-can-txqueuelen.rules
SUBSYSTEM=="net", ACTION=="add|change", KERNEL=="can*", ATTR{tx_queue_len}="10000"
EOF
```

marckleinebudde, 2022-07-02:
FTR: `systemd-networks` supports `txqueuelen`: https://www.freedesktop.org/software/systemd/man/systemd.link.html#TransmitQueueLength=

andrestoga, 2022-08-26:
> FTR: `systemd-networkd` supports `txqueuelen`: https://www.freedesktop.org/software/systemd/man/systemd.link.html#TransmitQueueLength=

Is this how you configure the txqueuelen in the network file?
TransmitQueueLength=1000

because it's not working for me

marckleinebudde, 2022-08-27:
It's a parameter in the .link file

mpena2099, 2022-08-29:
@marckleinebudde , so... I have to create a file named "80-can.link" and put "[Link] TransmitQueueLength=1000" on it, right?

But where exactly should this file be saved? I tried on "/etc/systemd/network" and "/usr/lib/systemd/network" and in both cases the file apparently didn't load.

marckleinebudde, 2022-09-05:
@mpena2099, you need some kind of `[Match]` section in the `.link` file.

andrestoga, 2022-09-19:
> 

I created a file named 80-can.link with the `[Match]` section in `/etc/systemd/network/` but it's not loading the configuration either:

```
[Match]
Name=can*

[Link]
TransmitQueueLength=1000
```

My network configuration:
```

[Match]
Name=can*

[CAN]
BitRate=500K
RestartSec=1ms
```

and the output of `networkctl status $INTERFACE`

```
Sep 19 15:42:58 pc systemd-networkd[2705]: can0: Lost carrier
Sep 19 15:42:58 pc systemd-networkd[2705]: can0: Gained carrier
Sep 19 15:42:59 pc systemd-networkd[2705]: can0: Lost carrier
Sep 19 15:42:59 pc systemd-networkd[2705]: can0: Gained carrier
Sep 19 15:43:00 pc systemd-networkd[2705]: can0: Lost carrier
Sep 19 15:43:00 pc systemd-networkd[2705]: can0: Gained carrier
Sep 19 15:43:01 pc systemd-networkd[2705]: can0: Lost carrier
Sep 19 15:43:01 pc systemd-networkd[2705]: can0: Gained carrier
Sep 19 15:43:09 pc systemd-networkd[2705]: can0: Lost carrier
Sep 19 15:43:09 pc systemd-networkd[2705]: can0: Gained carrier
```

marckleinebudde, 2022-09-19:
I think `Name` is not a valid `Match`, see https://man7.org/linux/man-pages/man5/systemd.link.5.html

marckleinebudde, 2022-09-19:
The output of `networkctl status $INTERFACE` looks incomplete. Which systemd version are you using?

```
$ networkctl status peakfd0 
● 15: peakfd0                                                    
                 Link File: /etc/systemd/network/70-usb-peak.link
              Network File: /etc/systemd/network/80-canfd.network
                      Type: can
                      Kind: can
                     State: carrier (configured)
              Online state: offline
                      Path: pci-0000:00:14.0-usb-0:1.1.4:1.0
                    Driver: peak_usb
                    Vendor: PEAK System
                     Model: PCAN-USB_FD
                       MTU: 72
                     QDisc: pfifo_fast
  Number of Queues (Tx/Rx): 1/1
         Activation Policy: up
       Required For Online: yes

Sep 18 22:55:37 hardanger systemd-networkd[400308]: peakfd0: Configuring with /etc/systemd/network/80-canfd.network.
Sep 18 22:55:37 hardanger systemd-networkd[400308]: peakfd0: Link UP
Sep 18 22:55:37 hardanger systemd-networkd[400308]: peakfd0: Gained carrier
Sep 19 08:55:08 hardanger systemd-networkd[400308]: peakfd0: Link DOWN
Sep 19 08:55:08 hardanger systemd-networkd[400308]: peakfd0: Lost carrier
Sep 19 08:55:08 hardanger systemd-networkd[400308]: peakfd0: Configuring with /etc/systemd/network/80-canfd.network.
Sep 19 08:55:08 hardanger systemd-networkd[400308]: peakfd0: Link UP
Sep 19 08:55:09 hardanger systemd-networkd[400308]: peakfd0: Gained carrier
```

andrestoga, 2022-09-19:
> The output of `networkctl status $INTERFACE` looks incomplete. Which systemd version are you using?
>
> ```
> $ networkctl status peakfd0
> ● 15: [...]

Ok, I added `OriginalName` instead and tried again:

```
~$ cat /etc/systemd/network/80-can.link
[Match]
OriginalName=can*

[Link]
TransmitQueueLength=1000
```

```
6: can0: <NOARP,UP,LOWER_UP,ECHO> mtu 16 qdisc pfifo_fast state UP group default qlen 10
    link/can
7: can1: <NOARP,UP,LOWER_UP,ECHO> mtu 16 qdisc pfifo_fast state UP group default qlen 10
    link/can
~$ networkctl status can0
● 6: can0
             Link File: /etc/systemd/network/80-can.link
          Network File: /etc/systemd/network/80-can.network
                  Type: can
                 State: carrier (configured)
                  Path: pci-0000:06:00.0
                Driver: advcan_pci_socket
                Vendor: Advantech Co. Ltd
                 Model: MIOe-3680 2-Port CAN-Bus MIOe Module with Isolation Protection
                   MTU: 16
  Queue Length (Tx/Rx): 1/1
     Activation Policy: up
   Required For Online: yes

Sep 19 17:44:39 rositapc3-UNO-2483G-474AE systemd-networkd[325]: can0: Link UP
Sep 19 17:44:39 rositapc3-UNO-2483G-474AE systemd-networkd[325]: can0: Gained carrier
~$ cat /etc/systemd/network/80-can.link
[Match]
OriginalName=can*

[Link]
TransmitQueueLength=1000
```

I'm still getting the `qlen=10` instead of `qlen=1000`

This is the systemd version I'm using:
```

~$ systemd --v
systemd 245 (245.4-4ubuntu3.17)
+PAM +AUDIT +SELINUX +IMA +APPARMOR +SMACK +SYSVINIT +UTMP +LIBCRYPTSETUP +GCRYPT +GNUTLS +ACL +XZ +LZ4 +SECCOMP +BLKID +ELFUTILS +KMOD +IDN2 -IDN +PCRE2 default-hierarchy=hybrid
```

mpena2099, 2022-09-19:
I still haven't been able to get it to work here. Same problem as @andrestoga .

marckleinebudde, 2022-09-19:
You need at least system 248. See https://github.com/systemd/systemd/blob/main/NEWS#L2386

RoadRunnr, 2022-10-08:
doesn't work, tested under Ubuntu 22.04.1 with systemd 2.49, txqueuelen on CAN ifs is not set.
My .link file:
```
[Match]
Name=can*

[Link]
TransmitQueueLength=1000
```

However, this is very likely a systemd bug and a can-tools problem.

marckleinebudde, 2022-10-10:
Hi @RoadRunnr,

what's the output of `networkctl status $INTERFACE`

djaniel, 2022-10-31:
Hi. 
I came here for help. I am working with a RPi and Ubuntu server 22. After adding file /etc/systemd/network/80-can.network and enabling systemd-networkd the CAN interface is UP after booting:

`
$ ip a
...
3: can0: <NOARP,UP,LOWER_UP,ECHO> mtu 16 qdisc pfifo_fast state UP group default qlen 10
    link/can 
`
Something I don't understand is that after enabling CAN, SSH access takes forever. I get a lot of:

`
ssh: connect to host my-host port 22: Connection refused
`
I have to wait several minutes to get access. I fear there are some configuration issues or this procedure is not the best way to do it. Probably related to netplan?
`
$ systemd --v
systemd 249 (249.11-0ubuntu3.6)
`

marckleinebudde, 2022-10-31:
Hey @djaniel,

I assume Ubuntu doesn't use `systemd-networkd` by default. I've never used `netplan` before, don't know if `systemd-networkd` and `netplan` can co-exist.

djaniel, 2022-10-31:
Thanks for your fast reply @marckleinebudde . I was just reading netplan generates configuration files for other daemons : https://netplan.io/ (1 min read)
But still, that does not explain my 'lag' issue. I will give Netplan a try.

ricardo-manriquez, 2022-11-26:
> Hi. I came here for help. I am working with a RPi and Ubuntu server 22. After adding file /etc/systemd/network/80-can.network and enabling systemd-networkd the CAN interface is UP after booting:
> 
> ```
> $ ip a
> ...
> 3: can0: <NOARP,UP,LOWER_UP,ECHO> mtu 16 qdisc pfifo_fast state UP group default qlen 10
>     link/can 
> ```
> 
> Something I don't understand is that after enabling CAN, SSH access takes forever. I get a lot of:
> 
> ```
> ssh: connect to host my-host port 22: Connection refused
> ```
> 
> When I look into the journal I can confirma a big lag:
> 
> ```
> $ journalctl -b
> Oct 31 09:19:53 my-host kernel: can: controller area network core
> [ a bunch of can related messages]
> Oct 31 09:22:00 my-host cloud-init[655]: ci-info: |  can0  |  True |
> [other messages]
> Oct 31 09:22:02 my-host sshd[823]: Server listening on 0.0.0.0 port 22.
> ```
> 
> As you can see I have to wait several minutes to get SSH access. I fear there are some configuration issues or this procedure is not the best way to do it. Probably related to netplan?
> 
> ```
> $ systemd --v
> systemd 249 (249.11-0ubuntu3.6)
> ```

I had the same issue, this is solved by adding a link stanza and configuring RequiredForOnline=no

The final configuration is as follows
File: /etc/systemd/network/80-can.network
```
[Match]
Name=can*

[Link]
RequiredForOnline=no

[CAN]
BitRate=500K
RestartSec=500ms
```

Of course the bitrate and restart policy are for my network
Best regards!

dan1229, 2023-05-23:
@marckleinebudde Thank you for this thread been really helpful, I am trying to set up `CAN-FD` and running into some issues though.

My config file at `/etc/systemd/network/80-can.network`:

```
[Match]
Name=can*

[CAN]
BitRate=1000000
DBitRate=5000000
FDMode=yes
Loopback=yes
```

Upon running `sudo systemctl restart systemd-networkd` everything looks good, it will update the `BitRate` when changed however `FDMode` doesn't seem to work? I've tried a bunch of different syntax and there isn't much online so curious if you've dealt with similar.

marckleinebudde, 2023-05-23:
Which systemd version are you using? Which CAN Controller are you using?

dan1229, 2023-05-23:
`systemctl restart systemd-networkd`

Using a Kvaser U100 - definitely supports FD, that's the only way I can get it to work normally I use this command to do it:

```
ip link set can0 type can bitrate 1000000 dbitrate 5000000 fd on
```

dan1229, 2023-05-23:
Sorry copied the wrong thing - `systemd 245 (245.4-4ubuntu3.20)`

marckleinebudde, 2023-05-23:
CANFD support was added in systemd-246

dan1229, 2023-05-23:
That is incredibly good to know wow thank you so much!

ChaosBlades, 2023-09-20:
80-can0.link does not change txqueuelen for me either. Had to use a UDEV rule.

`sudo nano /etc/udev/rules.d/99-persistent-network.rules
`

> SUBSYSTEM=="net", ACTION=="change|add", KERNEL=="can0"  ATTR{tx_queue_len}="128"

yuwata, 2024-01-16:
> I had the same issue, this is solved by adding a link stanza and configuring RequiredForOnline=no

After https://github.com/systemd/systemd/pull/30967 is merged, `RequiredForOnline=no` will not be necessary. If you'd like, please test the PR and comment on it. Thank you.

wingunder, 2024-02-01:
Hi,

I realize that this issue is closed, but maybe this could help someone.

> I'm using this file on my laptop and it works:
> 
> ```
> ➜ (pts/7) frogger@hardanger:~ (master) ✗ cat /etc/systemd/network/80-can.network 
> [Match]
> Name=can*
> 
> [CAN]
> BitRate=500K
> ```

I could not get the above to work (can-utils version: 2023.03-1) on Debian Trixie.
However, I found another solution that works for me, also using systemd voodoo:

`# cat /etc/systemd/system/set-can0-bitrate.service` 
```  INI
[Service]
Type=oneshot
ExecStart=/sbin/ip link set can0 down
ExecStart=/sbin/ip link set can0 type can bitrate 1000000
ExecStart=/sbin/ip link set can0 up

[Install]
WantedBy=sys-subsystem-net-devices-can0.device
```
I could not work out how to make this generic for all `can` devices, so for this solution, you'll need a one-shot service per device.

marckleinebudde, 2024-02-01:
There's already a proper tool for this, it's called `systemd-networkd`.

wingunder, 2024-02-02:
@marckleinebudde, thanks, I found out that it's not enough to just `systemctl daemon-reload` after editing `/etc/systemd/network/80-can.network`. I had to `systemctl restart systemd-networkd`, before the changes were seen by `systemd.network`. So, your solution works perfectly:
``` shell
cat /etc/systemd/network/80-can.network
````
``` INI
[Match]
Name=can*

[CAN]
BitRate=500K
```

yuwata, 2024-02-02:
> @marckleinebudde, thanks, I found out that it's not enough to just `systemctl daemon-reload` after editing `/etc/systemd/network/80-can.network`. I had to `systemctl restart systemd-networkd`, before the changes were seen by `systemd.network`. So, your solution works perfectly:

`networkctl reload` should also work.

wingunder, 2024-12-21:
I lately tried setting the `TransmitQueueLength` for a CAN device, and found a working solution here:
https://github.com/systemd/systemd/issues/30947#issuecomment-2510425738
Please note that I could only get the new `TransmitQueueLength` value to load, by physically rebooting. All attempts to restart or reload services, using `networkctl` and `systemctl` had no effect.

yuwata, 2024-12-21:
As replied at https://github.com/systemd/systemd/issues/30947#issuecomment-2558115535,
> @wingunder .link file is used by systemd-udevd, rather than systemd-networkd. So, after .link file is modified, you need to do the following to make the new value applied.
> 
> ```
> udevadm control --reload # or systemctl reload systemd-udevd.service
> udevadm trigger /sys/class/net/CAN_INTERFACE_NAME
> ```