linux-can/can-utils #71: J1939

Swompet, 2018-02-15:
Im trying to use this at a tractor.
Why dont I get an output here?
Candump brings me an output.

itachi1995, 2018-02-19:
you have to test it with oscilloscope

Swompet, 2018-02-21:
what do you mean? what can be the reason that candump works but cansniffer dont?

marckleinebudde, 2018-02-22:
What's the output of `candump`? Which program doesn't work? How do you start it?

Swompet, 2018-02-23:
can dump gives me an output like this:
1CE726A0#A87D69FFE60C0000
1CE6A026#A87D6900E60C0000
1CE726A0#A87E69FFE70C0000
1CE6A026#A87E6900E70C0000
1CE726A0#A87F69FFE80C0000
1CE6A026#A87F6900E80C0000
1CE726A0#A88069FFE90C0000
I using a raspberry pi with a Pican2.
and that the programs on the pi.
The problem might be the the 29 bit identifier on the isobus ISO 11783

coryjfowler, 2018-07-29:
Currently, the mainstream version of cansniffer does not support extended (29bit) IDs.  You would need to clone and make either [hartkopp's can-utils](https://github.com/hartkopp/can-utils) or [my can-utils](https://github.com/coryjfowler/can-utils).  I have not tested kartkopp's version as I was unaware of its existence until earlier this morning, but I can vouch that my version is currently working for me.

josemic, 2019-10-22:
Can this issue be closed now? Has hartkopp's can-utils been merged in the meantime? Are extended (29bit) IDs now supported?

hartkopp, 2019-10-23:
My attempt to make the cansniffer 29bit ID capable is still not finished. Maybe I wanted to much and need to restart that development.

closed 2019-12-06 by marckleinebudde