linux-can/can-utils #73: Detect Error frames with SocketCAN BCM

xatavian, 2018-02-20:
Hello,

I am trying to set up a C++ CAN system using SocketCAN BCM. While I can successfully detect data frames, I can't detect error frames. 

My method to read an incoming frame has the following code:
```c++

// ...
typedef struct {
        struct bcm_msg_head head;
        struct can_frame frame;
 } rx_bcm_msg;

while(true) {
        rx_bcm_msg bcm_frame;
        int bytesReceived = ::read(rxSocket_, &bcm_frame, sizeof(bcm_frame));

        if(bytesReceived <= 0)
                break;   

        cout << frame.head.can_id & CAN_ERR_FLAG << endl;
        cout << frame.frame.can_id & CAN_ERR_FLAG << endl;

        // ...
}

// ...
```
For my tests, there are no incoming or written can frames apart from the eventual error frames. 
Using a VCAN interface named vcan0, I try to activate a bus error by doing: 
```bash
> sudo ip link set vcan0 down
```
When doing so, I detect an incoming frame from the BCM but the output of the program is:
```
0
0
```

How can I detect error frames with BCM ?
Best regards.

hartkopp, 2018-02-20:
Hm. The BCM uses the optimized CAN ID filter lists for SFF/EFF frames and therefore defines the 'mask' value in the according CAN filter with this definition:
https://elixir.bootlin.com/linux/latest/source/net/can/bcm.c#L75

So it's not possible to filter for error message frames right now.

Although I never thought about this use-case it doesn't really fit to the BCM's approach to handle CAN IDs.
The BCM filters for data content changes in single given CAN IDs. But the CAN IDs in the error messages change based on the message type. You can't set a 'mask' on the BCM CAN filter separately. The mask is always set to 'all bits on'.

xatavian, 2018-02-20:
Thank you for the fast answer. 

From what you wrote, I understand that if I want to detect errors , I need to open a CAN_RAW socket somehow only letting error frames go through ?

hartkopp, 2018-02-20:
Yes. You can set a CAN filter e.g. to ONLY get error message frames in CAN_RAW.
In that case you can also filter for specific relevant bits in the CAN ID for the status updates that are relevant for you.
https://elixir.bootlin.com/linux/v4.9/source/Documentation/networking/can.txt#L521

xatavian, 2018-02-20:
Right. It is actually very straight forward. 

Thanks for your help !

xatavian, 2018-02-21:
Following my question, it turns out that a frame is still fired by the BCM when an error is detected. Is it expected behavior ? If it is the case, is it possible to disable the feature ?

hartkopp, 2018-02-21:
Can you give more details? The BCM should not be capable to receive error frames.

xatavian, 2018-02-21:
Turns out I was doing something wrong so nothing to see on this side: it is not that a frame is emitted, more than the socket is activated (or I think). 

```c++
#include <QCoreApplication>
#include <QSocketNotifier>
#include <QObject>
#include <QDebug>

#include <linux/can/bcm.h>
#include <linux/can.h>
#include <unistd.h>
#include <net/if.h>
#include <sys/ioctl.h>
#include <sys/fcntl.h>

typedef struct {
    struct bcm_msg_head head;
    struct can_frame frame;
 } rx_bcm_msg;

class CANWithBCM {
public:
    CANWithBCM(const char *itf): rxSocket_(-1) {

        struct ifreq rx_ifr;
        strcpy(rx_ifr.ifr_name, itf);

        if( (rxSocket_ = socket(PF_CAN, SOCK_DGRAM, CAN_BCM)) < 0) {
            qCritical() << "Error while creating the socket";
            return;
        }

        if(ioctl(rxSocket_ , SIOCGIFINDEX, &rx_ifr) < 0) {
            qCritical() << "Error while setting up the interface";
            return;
        }

        int rxFlags = ::fcntl(rxSocket_, F_GETFL, 0);
        if( ::fcntl(rxSocket_, F_SETFL, rxFlags | O_NONBLOCK) < 0 ) {
            qDebug() << "Error while setting up the RX socket as non-blocking";
        }

        struct sockaddr_can rx_addr;
        rx_addr.can_family = AF_CAN;
        rx_addr.can_ifindex = rx_ifr.ifr_ifindex;
        if(::connect(rxSocket_, (struct sockaddr*) &rx_addr, sizeof(rx_addr)) < 0) {
            qDebug() << "Error in RX socket connection";
            return;
        }

        bcmNotifier_ = new QSocketNotifier(rxSocket_, QSocketNotifier::Read);
        QObject::connect(bcmNotifier_, &QSocketNotifier::activated, [this](int /* socket */) {

            qDebug() << "Notified";

             while(true) {
                 rx_bcm_msg bcm_frame;
                 int bytesReceived = ::read(rxSocket_, &bcm_frame, sizeof(bcm_frame));

                 if(bytesReceived <= 0)
                     break;

                 qDebug() << "Got frame";
                 qDebug() << bcm_frame.head.can_id;
                 qDebug() << bcm_frame.frame.can_id;
             }

        });

        configure();
        qDebug() << "All set";
    }

    void configure() {

        rx_bcm_msg bcm_frame;
        can_frame mask;
            mask.can_dlc = 1;
            mask.can_id = 0x600;
            mask.data[0] = 0xFF;

        memset(&bcm_frame.head, 0x00, sizeof(bcm_msg_head));
        bcm_frame.head.opcode = RX_SETUP;
        bcm_frame.head.flags = RX_CHECK_DLC;
        bcm_frame.head.can_id = mask.can_id;
        bcm_frame.head.nframes = 1;
        memcpy(&bcm_frame.frame.data, mask.data, mask.can_dlc);

        if(::write(rxSocket_, &bcm_frame, sizeof(rx_bcm_msg)) < 0) {
            qDebug() << "Configure OK";
            return;
        }

    }

    ~CANWithBCM() {
        delete bcmNotifier_;
    }

private:
    int rxSocket_;
    QSocketNotifier *bcmNotifier_;
};

int main(int argc, char *argv[])
{
    QCoreApplication a(argc, argv);
    CANWithBCM canCtrl("vcan0");

    return a.exec();
}
```

Language used: C++11/Qt 5.3 on Ubuntu 16.04 LTS

```bash
> sudo modprobe vcan
> sudo ip link add vcan0 type vcan
> sudo ip link set vcan0 up
```

Launch the program and then,

```
> sudo ip link set vcan0 up
> sudo ip link set vcan0 down
```

The output of the program is then:
```
All set
Notified
```

It looks like the socket has been activated such that the QSocketNotifier fired the activated event. Is that expected behavior ?

Hope you don't have any problem with Qt :p

hartkopp, 2018-02-22:
`can_frame mask;
            mask.can_dlc = 1;
            mask.can_id = 0x600;
            mask.data[0] = 0xFF;`
Is the CAN frame 'mask' initialized properly?
What's the content of e.g. mask.data[1] ??

xatavian, 2018-02-22:
You're right, I forgot that part in my dummy code even though I don't think it really matters ...? Isn't the BCM supposed to only read data from the mask up to `mask.can_dlc` ?

Anyway, defining the mask as: 
```c++
can_frame mask;
memset(&mask.data, 0x00, sizeof(mask.data));
mask.can_dlc = 1;
mask.can_id = 0x600;
mask.data[0] = 0xFF; // Same behavior if I comment this line
```
gives the same result.

xatavian, 2018-02-22:
Taking a deeper look at the code, it turns out that `read`set errno as ENETDOWN. This might be related to the fact that I shut down the interface and therefore probably not linked to SocketCAN. Thanks for your help !

hartkopp, 2018-02-22:
In fact `mask.can_dlc` is not used.
See example at https://elixir.bootlin.com/linux/latest/source/Documentation/networking/can.txt#L838
Only the data[] content (and the **entire** 64 bit data[] content) is used from the struct can_frame's in the RX_SETUP message. The CAN ID to filter is defined in the bcm_msg_head.

closed 2018-03-16 by hartkopp