linux-can/can-utils #74: Candump showing 29 bit ID's instead of 11 bit - can't replay codes

physiii, 2018-02-21:
Trying to replay CAN codes for 2018 Dodge Promaster. From what I've read they use an 11 bit CAN ID. However when I use:

`candump -L can0 > can.log`

The log shows 29 bit ID's
```
(1519238827.107818) can0 04394100#00000000F0006000
(1519238827.122395) can0 063D4000#4652000080FE0000
(1519238827.126761) can0 04394000#000000AB
(1519238827.147822) can0 03029000#0000000000000000
(1519238827.152676) can0 06254000#00FFFF00FFFFFF00
(1519238827.157823) can0 03029100#0000000000000000
(1519238827.207842) can0 04394100#00000000F0006000
(1519238827.222727) can0 06314000#0000000000000000
(1519238827.226953) can0 04394000#000000AB
(1519238827.244633) can0 0E094000#001A00000409
(1519238827.247843) can0 03029000#0000000000000000
(1519238827.249614) can0 0E094003#001A
(1519238827.251375) can0 0E094020#000A
(1519238827.253121) can0 0E09400A#001A
(1519238827.254935) can0 0621400A#0000
(1519238827.257940) can0 03029100#0000000000000000
(1519238827.262128) can0 06314020#100000BC
```
I'm thinking this is why replaying with canplayer is not working. Does anyone know how I can force candump to use 11 bit ID's?

hartkopp, 2018-02-22:
candump just reflects what is received from the bus. When there are only 29 bit IDs you get the output you pasted above.
Your 2018 Dogde ProMaster is not 'a classical' Dogde but a re-labeled FIAT Ducato which was engineered in Italy and uses 29 bit IDs. That's why your referenced documentation does not fit.

physiii, 2018-02-22:
Okay 29 bit makes sense now - do you know why replaying dumped lock and unlock events does not lock or lock the vehicle?

hartkopp, 2018-02-22:
Probably you are looking at a status-only information and no controlling content. But this is nothing to be discussed as an issue of candump. Therefore I would like to close this issue if you don't mind.

closed 2018-02-22 by physiii