linux-can/can-utils #75: cansend. Valgrind reports points to uninitialised byte(s) in bind call.

Shmink, 2018-03-24:
```
==7553== Syscall param socketcall.bind(my_addr.sa_data) points to uninitialised byte(s)
==7553==    at 0x4F420F7: bind (syscall-template.S:84)
==7553==    by 0x400CBA: ??? (in /usr/bin/cansend)
==7553==    by 0x4E5A82F: (below main) (libc-start.c:291)
==7553==  Address 0xffefffd12 is on thread 1's stack
```

The program still works fine but in my own project it was an issue and I fixed it by adding `bzero(&addr,sizeof(struct sockaddr_can));` to zero out the struct. I tried to fork and add it myself but after doing that then running `make` and the command `./cansend vcan0 123#DEADBEEF` I get the error message `write: No such device or address`. Still learning C so not to sure why it's the case but your repo has been a big help to me with my project.

hartkopp, 2018-03-24:
Hi Tom,

assuming vcan0 is up and running at that time: At which line number did you put the bzero() call?

Regards,
Oliver

Shmink, 2018-03-24:
`vcan0` is up and running. It would have been line 138 in `cansend.c`.

hartkopp, 2018-03-24:
You should move it in front of line 105 ... otherwise you kill the settings made there.

Shmink, 2018-03-24:
Yep, that sorted it and the Valgrind message has now gone. I'll submit a pull request.

closed 2018-03-24 by Shmink