linux-can/can-utils #83: Should candump show the frames sent while the device is not connected to the bus?

gimohsent, 2018-05-16:
Hello. I'm using MPC2515 and it is not connected to a CAN bus. 
I set up the network like so:
`sudo ip link set can0 up type can bitrate 500000`
When I send a frame with the following command,
`cansend can0 000#DEADBEEF`
`candump can0` does not show it as an output.
But if I use the `loopback on ` option when setting up `can0`, `candump` produces two outputs, which, I assume, correspond to the frame send and the frame received.
```
  can0  000   [4]  DE AD BE EF
  can0  000   [4]  DE AD BE EF
```
Is this behaviour normal?

gimohsent, 2018-05-16:
Here's ifconfig's output after sending 12 frames, at which point the buffer becomes full.
```
can0      Link encap:UNSPEC  HWaddr 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00  
          UP RUNNING NOARP  MTU:16  Metric:1
          RX packets:10 errors:0 dropped:0 overruns:0 frame:0
          TX packets:10 errors:9 dropped:9 overruns:0 carrier:9
          collisions:0 txqueuelen:10 
          RX bytes:40 (40.0 B)  TX bytes:40 (40.0 B)
```

marckleinebudde, 2018-05-16:
Yes, `candump` will only show CAN frames that are actually send. Sending on a CAN bus means it received at least by one station on the bus.

closed 2018-05-16 by mohse-n