linux-can/can-utils #91: candump bridge issue, stops after 300 frames

Scytie, 2018-08-15:
I am running 3 Cantacts at a fully working car electronic from a VW Golf 7.
slcand, cansend and so on are fully working.
Now I want to create a bridge between two of the Cantacts them.
But if I ran `candump can0 -b can1`, it stops printing out the frames after exactly 300 Frames and freezes. I can not stop it than with Ctrl+C or Ctrl+D.

Do you have any Ideas why? In the code I cannot find a 300Frames limit.

hartkopp, 2018-08-15:
Can you check the queueing discipline for your CAN netdevice, whether you are using pfifo ?
See https://marc.info/?l=linux-can&m=152809682203669&w=2

Scytie, 2018-08-17:
Changing the queue with:
`tc qdisc replace dev can1 root pfifo`
Prevents candump from freezing. But candump ends with 'bridge write: No buffer space available'.
At the next start it stops immediately with 'bridge write: No buffer space available'.
Cansend isn't working than, too. Same Errormessage.

It seems that the frame is not being sended with changed queing. But if I candump the outgoing canbus, the packages are on the bus.

hartkopp, 2018-08-17:
IMO the main issue is the insufficient TX performance of slcan protocol devices (like CAN232, CANtact, etc). There are native USB CAN adapters e.g. from PEAK, 8devices, Kvaser and others that have complete CAN configuration support with the netlink interface (ip link ...) and with a better performance on the transmission side (see Linux kernel CAN driver config).
When the buffer runs full on the tx-side, you either send too much for the CAN bus or the bandwidth to the CAN controller is too small.
The bridging functionality of the candump-tool was a quick hack and solved easy use-cases. But we ran into the 'bridge write: No buffer space available' and frame drops very fast too, as all the CAN frames are routed via user-space (which is scheduled).
Did you try the can-gw with the cangw tool from can-utils?

hartkopp, 2018-10-24:
@Scytie : Any update on this? Otherwise I would like to close this issue.

closed 2018-10-27 by hartkopp

Scytie, 2018-10-29:
Sorry, i forgot to answer.
Using the CanDelight-Firmware: https://github.com/HubertD/candleLight_fw
fixes the Problem. It seems to be a problem with slcand and/or the original cantact firmware

hartkopp, 2018-10-29:
Thanks for the feedback!!
Good to know that @HubertD 's firmware can turn the cantact hardware into a powerful native Linux-CAN device, so that this solution can be recommend next time.

https://mobile.twitter.com/nixcraft/status/1054935148339363840
Just joking  :-)