linux-can/can-utils #94: [Question] Unsupported can bus speeds

asbachb, 2018-08-23:
Hi,

I currently try to communicate to a lowspeed can bus with a speed of 33kbps. Is there any way to start `slcand`with that speed? Is there any workaround?

Thanks
Benjamin

hartkopp, 2018-08-23:
Hi,
the numbers for the CAN bitrate are simply forwarded to the (SL)CAN adapter, e.g. the CAN232 from Lawicel http://www.can232.com/docs/can232_v3.pdf
When you look at the slcand option "-b <btr>" and look on the same page (6) in the above PDF, you can put any other raw bitrate configuration into the CAN232. But I don't know whether **your** adapter supports this config option too(!?!?).

With the tool can-calc-bittiming you can calculate the btr value (-> SJA1000).
You can find the example 0x31c for 125kBit/s from the PDF in the generated output too.
Just try: can-calc-bit-timing -b 125000 sja1000
can-calc-bit-timing -b 33000 sja1000 finally leads to 0x1517

Best regards,
Oliver

asbachb, 2018-08-26:
Thanks for your response. To be honest: I'm a little bit unsure how the canable adapter is implemented. I did a little but further investigation and found it that there's a alternative firmware (https://github.com/HubertD/candleLight_fw) without the need of using `slcand`. Maybe that's another way to go.

hartkopp, 2018-08-26:
Oh, yes! If you are able to use that alternative firmware from @HubertD , please do it :-)
That makes your adapter a real native Linux-CAN adapter which you can configure with the 'ip' tools.
Can you post your experienced here to help other users before I close this issue?

asbachb, 2018-08-26:
I upgrade my canable device and was able to set baudrates of 333000bps and 125000bps via `ip link set can0 up type can bitrate 333000` for example. I get messages from the different busses but did not find the message I was looking for. But basically I guess that's the way to go.

closed 2018-08-26 by asbachb