linux-can/can-utils #98: Issue with cangw 'modifying' b/w vcan0 / vcan1

psyolent, 2018-09-14:
Hi

Trying to use cangw to modify data seen on vcan0 and send on to vcan1.
eg:
cangw -A -s vcan0 -d vcan1 -f 551:3E -m SET:ILD:608.1.5300000000000000
(when you see 3E on PID 551 on vcan0, send it to vcan1 to PID 608, 1 byte as value 53)

the example works fine eg: cangw -A -s vcan0 -d vcan1 -e -f 551:C00007FF -m SET:IL:608.1.1122334455667788 but passes the exact value from the first byte on 551 to 608 ; i need to amend that value

i thought the SET did this? 

thoughts/ideas/very welcome. ta.

hartkopp, 2018-09-15:
The filter -f 551:3E is a canid:mask filter and not a canid:data,data,... filter.
So your assumption "when you see 3E" is not how it works. See the CAN filter description e.g. in candump -?

You can have multiple modifications on the received CAN frame (AND/OR/XOR/SET) where SET just sets the values given as parameter and AND/OR/XOR are the modifications that modify the incoming/original frame content.

psyolent, 2018-09-15:
thanks hartkopp for qualifying that for me. much appreciated.

closed 2018-09-16 by hartkopp