linux-can/can-utils #103: CANbus Errors Bring Down Multiple Sockets

nehrenholz, 2018-09-21:
When a CAN interface is disconnected from the bus by unplugging the hardware from the bus, and the socket connected to the unplugged CAN interface is continually sent data for transmission, all other sockets (including UDP) will be unable to receive/send data. The properly plugged in sockets will continue to operate for a certain amount of time, which seems to be proportional to the amount of data sent to the unplugged socket. This has been replicated with the peak_usb, sja1000, and kvaser_usb drivers, on Debian 8, 9 and Ubuntu. 

The CAN Tx/Rx, and UDP Rx are continually looped to retry when errNo is EINTR, similar to:
`do
{
	n = read( fd, buf, CAN_FRAME_SIZE );
	if ( errNo == EINTR )
        {
            retry = TRUE;           // Retry for an interrupted call.
        }
} while ( retry );`

UDP Rx is similar however uses recvfrom() in the loop. The CAN socket is setup as follows:
`// ---- Create a socket ----
    fd = socket( PF_CAN, sockType, protocol );
// ---- Get the CAN interface index ----
    strncpy( ifr.ifr_name, if, CAN_NAME_SIZE );
    ioctl( fd, SIOCGIFINDEX, &ifr );
    addr.can_family  = AF_CAN;
    addr.can_ifindex = ifr.ifr_ifindex;
    bind( fd, (RawSocketAddr *) &addr, sizeof( addr ) );`

hartkopp, 2018-10-03:
If you have different CAN_RAW sockets that are bound to a single interface each (and not to 'any') and you unplug that CAN hardware only this socket will return an error code (IIRC: -ENODEV). And from this time it makes no sense to send any data to this specific socket. Other sockets should not be affected on this single interface.

Please try to play with cangen/candump first. I assume your loop with checking for -EINTR causes the problem.

Regards,
Oliver

hartkopp, 2018-10-24:
@nehrenholz : Any update on this?

closed 2018-10-27 by hartkopp