linux-can/can-utils #104: candump -B can1 can0 fails when CAN interfaces are in CAN-FD mode.

jm3lee, 2018-10-02:
Here's a prototype fix against 1c81f7f. This won't work if interfaces are in CAN 2.0 mode.

```
diff --git a/candump.c b/candump.c
index f98d1c3..4217f2a 100644
--- a/candump.c
+++ b/candump.c
@@ -332,6 +332,8 @@ int main(int argc, char **argv)
                                                   &loopback, sizeof(loopback));
                                }
 
+                setsockopt(bridge, SOL_CAN_RAW, CAN_RAW_FD_FRAMES, &canfd_on, sizeof(canfd_on));
+
                                if (bind(bridge, (struct sockaddr *)&addr, sizeof(addr)) < 0) {
                                        perror("bridge bind");
                                        return 1;
```

hartkopp, 2018-10-03:
You are right!
Would you like to provide a proper pull request (e.g. "candump: enable bridging function for CAN FD interfaces") with correct white spaces (tabs)?!?
Or do you want me to make it a patch and mention you as reporter of that issue?

Btw. the bridging function was some kind of quick hack too. If you want to route (and modify) CAN frames between CAN interfaces you should try the cangw tool and the can-gw kernel module.

Regards,
Oliver

hartkopp, 2018-10-24:
Fixed in commit https://github.com/linux-can/can-utils/commit/c45a17e96ba307ad752ac4551513559daefbefbd 
Thanks!

closed 2018-10-24 by hartkopp