linux-can/can-utils #110: slcand and CAN-FD

umlaeute, 2018-10-17:
after i finally got my [`USB-CAN ISO`](http://www.vscom.de/vscom-usb-can-iso.html)  working with `slcand`, i noticed that it only supports `CAN` but not `CAN-FD` (with 29-bit identifiers).

~~~
$ ip link show slcan0
18: slcan0: <NOARP,UP,LOWER_UP> mtu 16 qdisc pfifo_fast state UNKNOWN mode DEFAULT group default qlen 10
    link/can
~~~

Is there a way to change the MTU to `72` to receive CAN-FD frames?
(to be honest, I don't even know whether my device supports CAN-FD frames; but i figure that `slcand` does not)

marckleinebudde, 2018-10-17:
`CAN-FD` and 29-bit identifiers are two pairs of shoes. However, AFAICS the linked usb device doesn't support `CAN-FD`:
> Supports CAN 2.0A and CAN 2.0B
> CAN High Speed up to 1 MBit/s

yegorich, 2018-10-17:
Is `CAN-FD` defined for `slcan` at all?

umlaeute, 2018-10-17:
yep, i just noticed that `CAN-FD` and 29bit identifiers (`EFF`) are two separate things (admittedly, i'm rather new to CAN)

`EFF` (29bit IDs) works fine on the device (at least, it works as well as "normal" CAN messages; which is not very well, but might be a problem with the device)

`CAN-FD` does not at all (and this might be a problem with the device, as @marckleinebudde  indicated).

however, afaict @yegorich is right and there is no `CAN-FD` built into `slancd` in the first place.

hartkopp, 2018-10-19:
AFAIK there's no SLCAN device that supports CAN FD by now. The original manufacturer Lawicel of CAN232 and CANUSB (http://www.canusb.com/) created the SLCAN ASCII protocol which doesn't support CAN FD. If you need a CAN FD USB device with Linux Mainline Support you might better use native CAN FD USB adapter from PEAK System or Kvaser.

umlaeute, 2018-10-19:
thanks @hartkopp for the clarification. if the underlying protocol (SLCAN ASCII) does not support `CAN-FD`, then it's no wonder `slcand` doesn't support it.

thanks for being patient.

closed 2018-10-19 by umlaeute