linux-can/can-utils #112: candump logging with absolute date and time

travlytle, 2018-12-06:
While using candump to create log files I want date and time.  As see below, only when I dump to console do I get full date and time.  The log file is only getting time. 

I'm not sure if this has to do with my hardware or system.  I've tested on:
1) linux in virtualbox using PCAN using (2018.02.0-1).
2) raspberry pi with PiCAN using using (0.0+git20161220-1)
3) telematics device built using buildroot, I'm not able to find the version, but it states:
    CAN_UTILS_VERSION = c3305fdd515464153d20199db232b6124bc962c0
    CAN_UTILS_SITE = $(call github,linux-can,can-utils,$(CAN_UTILS_VERSION))
    CAN_UTILS_LICENSE = BSD-3-Clause or GPL-2.0, GPL-2.0+
    CAN_UTILS_AUTORECONF = YES

```
trav@svbox:~$ candump -tA -l can0
Disabled standard output while logging.

Enabling Logfile 'candump-2018-12-06_093119.log'

^Ctrav@svbox:~$ cat candump-2018-12-06_093119.log 
(1544106679.432583) can0 562#122DF953813CA2
(1544106679.631330) can0 1F0#4C96281B62ACBC20
(1544106679.830333) can0 687#C3F26E
(1544106680.029125) can0 412#06C136
(1544106680.228086) can0 67E#01C65274D19FAC6D
(1544106680.427065) can0 4FA#5F6502615110B55A
(1544106680.625958) can0 429#6FB48932FBCECE26
trav@svbox:~$ candump -tA can0
 (2018-12-06 09:31:30.572475)  can0  57E   [5]  C7 35 06 18 00
 (2018-12-06 09:31:30.770509)  can0  03C   [4]  4E DF FC 32
 (2018-12-06 09:31:30.969856)  can0  56F   [8]  98 38 7E 6E DA E3 B0 7F
 (2018-12-06 09:31:31.168410)  can0  23B   [8]  28 43 84 0C C7 00 DD 68
 (2018-12-06 09:31:31.367523)  can0  59A   [3]  A3 D4 11
 (2018-12-06 09:31:31.566226)  can0  5EB   [8]  CA AA D3 27 C6 72 96 29
 (2018-12-06 09:31:31.765056)  can0  793   [1]  59
 (2018-12-06 09:31:31.963989)  can0  0C2   [3]  35 02 DD
^Ctrav@svbox:~$
```

travlytle, 2018-12-06:
After review, I think what I'm asking doesn't make sense because the timestamp is there in epoch format.  What I'm realizing is when I convert the log using log2asc the timestamp is lost.  Then I'm further processing the asc file into blf, thus no timestamp there either.

hartkopp, 2018-12-07:
What do you mean with " when I convert the log using log2asc the timestamp is lost" ??
Do you expect an issue with log2asc in your setup?

travlytle, 2018-12-07:
I'd have to go back and check, but I believe the log2asc keeps the timestamp differently, thus I thought it was lost.  I believe it puts a timestamp at the top of the file, then the messages timestamps have to be calculated from there?

I also realized that the software I was using to read the blf had two ways of playing back the file.  
1) in a replay block which timestamps start from start of playback
2) offline mode which shows the recorded timestamps (which is what I wanted)

hartkopp, 2019-08-29:
> I'd have to go back and check, but I believe the log2asc keeps the timestamp differently, thus I thought it was lost. I believe it puts a timestamp at the top of the file, then the messages timestamps have to be calculated from there?

Yes, that is ASC file style. :-)

Closing again.

closed 2019-08-29 by hartkopp