linux-can/can-utils #115: can-gw ; help ORing and XORing

psyolent, 2018-12-19:
Just after a little help with my understanding of incrementing a value. I have a value which i've masked with FF (tach value) which comes on PID 551. The value changes (as a tach does) however while the static OR of FF to the value that comes in allows me to the XOR it to the value I need ; as the incoming value changes so does that of the XOR. 

In the case of a value 3F hex coming in, if I OR it with FF then XOR it with AB it becomes hex 54 which is what I need. With the next value hex 40 however, then I OR it with FF my XOR value changes to 85 to make hex 55 which is what i need. 

The command as below is what I have come up with. I've not executed it as my test vehicle is some distance away which is annoying. I'm only working on the logic.

cangw -A -e -s can0 -d can1 -m -f 551:FF -m OR:D:608:FF00000000000000 -m XOR:D:608:AB00000000000000

Advice very welcomed. Ta.

hartkopp, 2018-12-19:
I assume you did not take into account that the **order of modifications** is fixed inside the can-gw kernel module:

https://github.com/linux-can/can-utils/blob/master/cangw.c#L219
`The max. four modifications are performed in the order AND -> OR -> XOR -> SET`

If you have different CAN-IDs you want to modify I would suggest to create multiple cangw jobs.

Additionally the filter `180:FF` looks incomplete. If you want to filter for a single SFF frame the mask should look like `-f 180:C00007FF` (see `candump`help text).

psyolent, 2018-12-20:
--> I assume you did not take into account that the order of modifications is fixed inside the can-gw kernel module:
dammit, i have my SET at the start - thanks for pointing that out mate. ta.

--> If you have different CAN-IDs you want to modify I would suggest to create multiple cangw jobs.
yes for the different PIDs different rules - no problems. 

--> Additionally the filter 180:FF looks incomplete. If you want to filter for a single SFF frame the mask should look like -f 180:C00007FF (see candumphelp text).
ok. will do more research. 

thanks for your input and help - very appreciated

closed 2019-01-19 by hartkopp