linux-can/can-utils #124: Problems when using extended and normal addressing in the same process

hfn92, 2019-01-19:
It seems that once a can socket has been created with extended addressing all subsequently created sockets are also created with extended addressing even if not intended.

I created a minimal example to reproduce this behavior. This program creates a regular 11bit listener (0x600), a listener with extaddr (0x800) and after that another 11bit listener (0x601). I essentially just copied the code from isotprecv for this example.

- Start the example `./ExtAddrTest`
- In another bash run 
```
echo 11 | isotpsend -s 600 -d 700 vcan0 ; \
echo 22 | isotpsend -s 600 -d 700 vcan0 ; \
echo 33 | isotpsend -x 44:33 -s 800 -d 900 vcan0 ; \
echo 44 | isotpsend -s 601 -d 701 vcan0 ; \
echo 55 | isotpsend -s 601 -d 701 vcan0 
```
-  Output of ExtAddrTest
```
00000600 11 
00000600 22 
00000800 33 
```

Also once I send using extaddr, ALL of the sockets seem to switch to extaddr even the ones create before the extaddr channel. This behavior is not reproduced by the example.

[ExtAddrTest.zip](https://github.com/linux-can/can-utils/files/2775400/ExtAddrTest.zip)

hartkopp, 2019-01-19:
When I invoke the isotpsend commands as provided by you, I can see this output with `candump`:

```
$ candump -ta any
 (1547896318.129047)  vcan0  600   [2]  01 11
 (1547896336.386195)  vcan0  600   [2]  01 22
 (1547896345.905141)  vcan0  000   [3]  44 01 33
 (1547896366.315357)  vcan0  601   [2]  01 44
 (1547896384.926886)  vcan0  601   [2]  01 55
```
The first thing that catches the eye immediately that there's a 11 bit CAN-ID 0x800, which doesn't fit and is sanitzed to 0x000. The reset with the `000   [3]  44 01 33` looks good for the requested extended addressing.

Can you please fix the example (with the 0x800, 0x900) and if it still doesn't work make a try with `isotprecv` whether the behaviour is what you would expect?

hfn92, 2019-01-19:
Sorry mistake on my part. I changed the ids to 0x100 and 0x200. But the problem persists.

[ExtAddrTestFixed.zip](https://github.com/linux-can/can-utils/files/2775573/ExtAddrTestFixed.zip)

I was not able to reproduce it with `isotprecv` alone since it only happens when creating multiple sockets in the same process. `isotprecv` alone works fine.

I added log messages to `can-isotp` 
```
static int isotp_bind(struct socket *sock, struct sockaddr *uaddr, int len)
{
      [...]

	int ae = (so->opt.flags & CAN_ISOTP_EXTEND_ADDR)? 1:0;
	printk("isotp_bind %X - EXT = %i\n", addr->can_addr.tp.rx_id, ae);
      [...]
```
and after running `ExtAddrTest` `dmesg` shows:
```
[30771.364199] isotp_bind 600 - EXT = 0
[30771.364286] isotp_bind 100 - EXT = 1
[30771.364361] isotp_bind 601 - EXT = 1
```

hartkopp, 2019-01-19:
I'm not that sure about threads here.  Are you sure you are creating 3 different sockets or is it just one socket that is shared or at least 'copied' as you would have it with fork()?

Did you ever try `strace ./ExtAddrTest` ?

I grep'ed the output for 'sock' and got this:
```
$ cat strace.txt | grep sock
socket(AF_CAN, SOCK_DGRAM, 6)           = 3
setsockopt(3, 0x6a /* SOL_?? */, 1, "\0\0\0\0\0\0\0\0\0\0\0\0", 12) = 0
setsockopt(3, 0x6a /* SOL_?? */, 2, "\0\0\0", 3) = 0
socket(AF_UNIX, SOCK_DGRAM|SOCK_CLOEXEC, 0) = 4
socket(AF_CAN, SOCK_DGRAM, 6)           = 4
setsockopt(4, 0x6a /* SOL_?? */, 1, "\2\2\0\0\0\0\0\0003\0\0D", 12) = 0
setsockopt(4, 0x6a /* SOL_?? */, 2, "\0\0\0", 3) = 0
socket(AF_UNIX, SOCK_DGRAM|SOCK_CLOEXEC, 0) = 5
socket(AF_CAN, SOCK_DGRAM, 6)           = 5
setsockopt(5, 0x6a /* SOL_?? */, 1, "\2\2\0\0\0\0\0\0003\0\0D", 12) = 0
setsockopt(5, 0x6a /* SOL_?? */, 2, "\0\0\0", 3) = 0
socket(AF_UNIX, SOCK_DGRAM|SOCK_CLOEXEC, 0) = 6
```
Obviously your C++ /threaded approach sets the sockopt() two times as you correctly observed it.

So it occurs somewhere in your user space application - and not in isotp.c ;-)

hfn92, 2019-01-19:
As long as `socket(PF_CAN, SOCK_DGRAM, CAN_ISOTP)` creates a different socket each time I'm creating 3 sockets.

I'm not sure what you mean by calling sockopt() two times. I call it two times just like in in `isotprecv`

https://github.com/linux-can/can-utils/blob/2328c6cfa0b004173644796a3604392f355dcde2/isotprecv.c#L220
https://github.com/linux-can/can-utils/blob/2328c6cfa0b004173644796a3604392f355dcde2/isotprecv.c#L221

hartkopp, 2019-01-19:
Right. But as you can see in the marked lines here:
``` 
$ cat strace.txt | grep sock
(..)
socket(AF_CAN, SOCK_DGRAM, 6)           = 4
--> setsockopt(4, 0x6a /* SOL_?? */, 1, "\2\2\0\0\0\0\0\0003\0\0D", 12) = 0
(..)
socket(AF_CAN, SOCK_DGRAM, 6)           = 5
--> setsockopt(5, 0x6a /* SOL_?? */, 1, "\2\2\0\0\0\0\0\0003\0\0D", 12) = 0
```
the last two sockets both set the extended addressing. Which is what your program does.

I did a `strace -x` which shows the extended address values in hex:
```
$ cat strace.txt | grep sock
socket(AF_CAN, SOCK_DGRAM, 6)           = 3
--> setsockopt(3, 0x6a /* SOL_?? */, 1, "\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00", 12) = 0
setsockopt(3, 0x6a /* SOL_?? */, 2, "\x00\x00\x00", 3) = 0
socket(AF_UNIX, SOCK_DGRAM|SOCK_CLOEXEC, 0) = 4
socket(AF_CAN, SOCK_DGRAM, 6)           = 4
--> setsockopt(4, 0x6a /* SOL_?? */, 1, "\x02\x02\x00\x00\x00\x00\x00\x00\x33\x00\x00\x44", 12) = 0
setsockopt(4, 0x6a /* SOL_?? */, 2, "\x00\x00\x00", 3) = 0
socket(AF_UNIX, SOCK_DGRAM|SOCK_CLOEXEC, 0) = 5
socket(AF_CAN, SOCK_DGRAM, 6)           = 5
--> setsockopt(5, 0x6a /* SOL_?? */, 1, "\x02\x02\x00\x00\x00\x00\x00\x00\x33\x00\x00\x44", 12) = 0
setsockopt(5, 0x6a /* SOL_?? */, 2, "\x00\x00\x00", 3) = 0
socket(AF_UNIX, SOCK_DGRAM|SOCK_CLOEXEC, 0) = 6
```

hfn92, 2019-01-19:
Ok this is embarrassing. I never noticed that the socket options were declared as static.
```
static struct can_isotp_options opts;
static struct can_isotp_fc_options fcopts;
static struct can_isotp_ll_options llopts; 
```
I'm so sorry for wasting your time!

closed 2019-01-19 by hfn92

hartkopp, 2019-01-20:
No problem - at least we found the issue. That's SW development ;-)