linux-can/can-utils #126: Add 2 examples for sending the same can frame with can_id as SFF and with EFF

jogo-, 2019-02-13:
https://github.com/linux-can/can-utils/blob/2328c6cfa0b004173644796a3604392f355dcde2/cansend.c#L85
e.g. 7FF#05FF / 000007FF#0EFF
This example should be clear enough and could help.

hartkopp, 2019-02-13:
There are two lines holding Classic CAN / CAN FD and '.' separated content for SFF/EFF examples.
I don't get your idea for an improvement.

jogo-, 2019-02-13:
It's just to clarify how to send the same frame with the same id as SFF or as EFF

hartkopp, 2019-02-13:
What about this as an example?
```
e.g. 5A1#11.2233.44556677.88 / 123#DEADBEEF / 5AA# / 123##1 / 213##311223344
     1F334455#1122334455667788 / 123#R / 00000123#R3
```

jogo-, 2019-02-13:
Ok, but:
`e.g. 5A1#11.2233.44556677.88 / 123#DEADBEEF / 5AA# / 123##1 / 213##311223344
     1F334455#1122334455667788 / 123#R / 7FF#05FF / 000007FF#0EFF`
is a funny way to clarify things using hexspeak (as in DEADBEEF) ;-)

hartkopp, 2019-02-14:
Reading source code must be fun :-D

closed 2019-02-14 by hartkopp