linux-can/can-utils #127: How to resolve the following error? scan0: ERROR while getting interface flags: No such device

Prriyamvradha, 2019-02-13:
I used the following commands in SocketCAN and I used the CANtact toolkit as a serial bus interface.ALL the commands work fine except the last command it throws the following error 
scan0: ERROR while getting interface flags: No such device.
OS-Ubuntu 18.04.1 

`sudo modprobe can
sudo modprobe can_raw
sudo modprobe slcan`

`slcand -o -s6 -t hw -S 3000000 /dev/ttyACM0 slcan0`

`ifconfig slcan0 up`

hartkopp, 2019-02-13:
Hm `slcan0` is the standard name and you are renaming this to `slcan0`.
Can you test, if you get the same error when using `can0` instead of `slcan0` in the last two lines?

Prriyamvradha, 2019-02-13:
> Hm `slcan0` is the standard name and you are renaming this to `slcan0`.
> Can you test, if you get the same error when using `can0` instead of `slcan0` in the last two lines?

Yes, I get the same error.

hartkopp, 2019-02-13:
I tried it on my own. You need to be `root` to run `slcand`.

Please try `sudo slcand -o -s6 -t hw -S 3000000 /dev/ttyACM0 slcan0` and `sudo ifconfig slcan0 up`

josemic, 2020-06-30:
@Prriyamvradha:
Is ithis solved? Can we close this issue?

miimoza, 2021-03-11:
I get the same problem:

`sudo slcand -o -c -s1 /dev/ttyACM0 can2`
`sudo ifconfig can2 up`

`can2: ERROR while getting interface flags: No such device`