linux-can/can-utils #128: candump shows no data

theGeekyLad, 2019-03-03:
I've configured my MCP2515 CAN bus module which works well with candump in loopback mode. Otherwise, it displays no traffic from the CAN bus. I've tried the setup on multiple cars but zero results.

I'm using the last version of Raspbian on my Raspberry Pi 3B+ with a 4.14.79-v7+ kernel.

hartkopp, 2019-03-03:
What do you mean with
> candump in loopback mode
??
Did you configure the MCP2515 in loopback mode?
What does ``ip -det link show can0`` show on your system?

theGeekyLad, 2019-03-03:
Yes, configuting the MCP2515 in loopback mode using `sudo ip link set can0 up type can bitrate 125000 loopback on` works fine with two terminals open, one that does a `cansend` and the other with `candump` running.

The same doesn't work without loopback, i.e. on the actual CAN bus in my car - no results with `candump` command issued.

Here's the result of `ip -det link show can0`,
```
4: can0: <NOARP,ECHO> mtu 16 qdisc noop state DOWN mode DEFAULT group default qlen 10
    link/can  promiscuity 0 
    can state STOPPED restart-ms 0 
	  mcp251x: tseg1 3..16 tseg2 2..8 sjw 1..4 brp 1..64 brp-inc 1
	  clock 4000000numtxqueues 1 numrxqueues 1 gso_max_size 65536 gso_max_segs 65535
```

Thanks for replying!

hartkopp, 2019-03-03:
The state of the CAN interface is not 'up' and no bitrate is visible ...

theGeekyLad, 2019-03-03:
> The state of the CAN interface is not 'up' and no bitrate is visible ...

Have a look at this blog - https://harrisonsand.com/can-on-the-raspberry-pi/. He's got a similar output, the only difference being the results which he's getting and I'm not. It's working fine in loopback. Is it a kernel related issue?

hartkopp, 2019-03-03:
Looks like you did the ``sudo ip link set can0 down``
When the interface is down, there is no traffic.
You should omit that line.

theGeekyLad, 2019-03-04:
For the same reason I didn't issue that command before running `candump` but still no go. I tried the same setup on another vehicle last night but it stands the same.

hartkopp, 2019-03-04:
But this doesn't fit to the output of ``ip -det link show can0`` you posted here https://github.com/linux-can/can-utils/issues/128#issuecomment-469042110

As the output indicates the interface is down.

theGeekyLad, 2019-03-04:
I hadn't set the link up before issuing the command you'd asked me to though. Without the interface being up, `candump` itself says *"network is down."* which isn't the case here. With the interface up and `candump` running, I get no traffic.

hartkopp, 2019-03-04:
Can you please send the result of ``ip -det link show can0`` when the interface is set to ``up``?

snoop911, 2019-05-30:
I'm having a very similar issue..

Using 'cansend can0' to transmit data, I verify with a scope that CAN-H and CAN-L have traffic and it's at the expected bit-rate (100kbps).. however, candump doesn't see any of this traffic.  I don't have anything on the bus except for the Pi / Can Hat that is doing the transmitting, so there isn't any acknowledge  (I'm just trying to log what I send).

I think the problem has to do with CAN_RAW_RECV_OWN_MSGS, doesn't seem to ever get set:
CAN_RAW_RECV_OWN_MSGS is OFF
CAN_RAW_LOOPBACK is ON

This is the how I'm setting it (kernel 4.14.98+):

if (setsockopt(s, SOL_CAN_RAW, CAN_RAW_RECV_OWN_MSGS , &enable_canfd, sizeof(enable_canfd))){
printf("error when enabling RECV_OWN_MSGS support\n");
return 1;
}

if(getsockopt(s, SOL_SOCKET, CAN_RAW_RECV_OWN_MSGS, &optval, &optlen) < 0) {
perror("getsockopt()");
close(s);
exit(EXIT_FAILURE);
}

printf("CAN_RAW_RECV_OWN_MSGS is %s\n", (optval ? "ON" : "OFF"));  //***********

if (setsockopt(s, SOL_CAN_RAW, CAN_RAW_LOOPBACK , &enable_canfd, sizeof(enable_canfd))){
printf("error when enabling RAW_LOOPBACK support\n");
return 1;
}

if(getsockopt(s, SOL_SOCKET, CAN_RAW_LOOPBACK, &optval, &optlen) < 0) {
perror("getsockopt()");
close(s);
exit(EXIT_FAILURE);
}

printf("CAN_RAW_LOOPBACK is %s\n", (optval ? "ON" : "OFF"))   //***********
if (setsockopt(s, SOL_CAN_RAW, CAN_RAW_FD_FRAMES, &enable_canfd, sizeof(enable_canfd))){
printf("error when enabling CAN FD support\n");
return 1;
}

ip -det link show can0
3: can0: <NOARP,UP,LOWER_UP,ECHO> mtu 16 qdisc pfifo_fast state UNKNOWN mode DEFAULT group default qlen 10
    link/can  promiscuity 0
    can state ERROR-PASSIVE restart-ms 0
          bitrate 100000 sample-point 0.850
          tq 500 prop-seg 8 phase-seg1 8 phase-seg2 3 sjw 1
          mcp251x: tseg1 3..16 tseg2 2..8 sjw 1..4 brp 1..64 brp-inc 1
          clock 4000000numtxqueues 1 gso_max_size 65536 gso_max_segs 65535

hartkopp, 2019-05-30:
You only see 'sent' CAN frames, when they have be sent successfully on the CAN bus. If you don't have a second CAN node that acks your sent frame, you did not perform a successful transmission - and therefore you don't see that frame. CAN_RAW_RECV_OWN_MSGS refers only to the socket that sent the frame. When you are using a separate candump instance there's not need for CAN_RAW_RECV_OWN_MSGS.

DecentBumpkin, 2019-06-10:
I have exactly the same problem,with loopback on I can receive some data using candump, but cannot read any of the traffic when loopback turned off. So terrible! I am brand new to canbus and closedly followed the tutorials but the result is terrible.
4: can1: <NOARP,UP,LOWER_UP,ECHO> mtu 16 qdisc pfifo_fast state UNKNOWN mode DEFAULT group default qlen 1000
    link/can  promiscuity 0 
    can state ERROR-ACTIVE restart-ms 0 
	  bitrate 1000000 sample-point 0.750 
	  tq 125 prop-seg 2 phase-seg1 3 phase-seg2 2 sjw 1
	  mcp251x: tseg1 3..16 tseg2 2..8 sjw 1..4 brp 1..64 brp-inc 1
	  clock 8000000

Prasad454, 2019-08-29:
Even i face similar problem, but not with loopback mode, i cant see any data on candump when i connect two nodes, as we have any alternative to test CAN traffic with one node and another node on a simulator? As we have shortages of HW, it difficult for me test with 2 nodes always.

Please find the below 

1st node
[image]

2nd node
[image]

marckleinebudde, 2019-08-29:
Your CAN bus is probably faulty. For a working CAN setup, you need at least two nodes, the same bitrate on all nodes and proper bus termination. Two nodes can be `can0` and `can1` of the same board though.
For the `cansend` `candump` example above, you have to start `candump` _before_ you send any data.

closed 2019-12-06 by marckleinebudde

EmadDeve20, 2021-11-23:
Hi Guys!
i read all of post but no any post is help me!

my problem is after this ```ip link set can0 up``` command, my terminal got freeze!

my device is orangepi +2e and my os is armbian

marckleinebudde, 2021-11-23:
Hey @EmadDeve20. The original problem was about a mcp2515 CAN on a raspi. What about opening a new issue for your problem. Please add which CAN adapter you're using.

EmadDeve20, 2021-11-23:
hi @marckleinebudde you are right i have to create an issue for orange pi and i'm sorry about this :D