linux-can/can-utils #133: Not Receiving or Receiving Wrong Messages

mwrouse, 2019-03-25:
Hello, 

I am having an issue with receiving CAN messages sent from an Arduino (using [this MCP2515 library](https://github.com/autowp/arduino-mcp2515)) on a Raspberry Pi. The Arduinos have no problem sending and receiving to one another.  The bus is terminated on both ends using 120Ω resistors between CAN_H and CAN_L.

I have followed several tutorials I've found on Google ([this](https://harrisonsand.com/can-on-the-raspberry-pi/), [this](https://www.hackster.io/youness/how-to-connect-raspberry-pi-to-can-bus-b60235), or [this](https://vimtut0r.com/2017/01/17/can-bus-with-raspberry-pi-howtoquickstart-mcp2515-kernel-4-4-x/)), and am having no luck.

I am using the eBay MCP2515 modules, powered by 3.3V (on both Pi and Arduino), I've verified that CAN_H and CAN_L have continuity to the module.

My problem: the MCP2515 on the Arduinos fail (if only one of them and the raspberry pi) to transmit, if I add in another they start transmitting successful again. When just one Arduino and Pi I receive garbage data with `candump can0`.

I've verified the module connected to the Pi works by using it with an Arduino. 

I am not familiar enough with Linux to go about this on my own anymore, and have been unable to find a solution online. So if someone here could assist, tell me how to get logs, etc, that would be fantastic.

In my `/boot/config.txt` file I have: 
```
dtparam=spi=on
dtoverlay=mcp2515-can0,oscillator=8000000,interrupt=25
#dtoverlay=spi0-hw-cs
```
I've tried several values for the second `dtoverlay` but none have worked, and I read somewhere that it isn't needed, so that's why it is commented out. 

If I start it in loopback mode (`sudo ip link set can0 up type can bitrate 500000 loopback on`)
And transmit using: 
```
cansend can0 5A1#11.22.33.44.55.66.77.88
```

I see this with `candump can0`:
```
  can0  5A1   [8]  11 22 33 44 55 66 77 88
  can0  5A1   [8]  11 22 33 44 55 66 77 88
```
(It actually shows up twice).

The Arduinos are configured for 500K bit rate, and to use 8MHz crystal.

The garbage data I am referring to shows up as something with the wrong ID and a length of `[0]` in candump:
```
  can0  13E2F907   [0]
  can0  18BE41DF   [0]
  can0  1C83BE9F   [0]
  can0  0EFA7E0F   [0]
  can0  1A7E0F1F   [0]
```
There is just a steady stream of those messages when in this state.

`uname -a` says I am on kernel 4.14.98+

hartkopp, 2019-03-25:
It makes sense, that the CAN frame shows up twice when you set `loopback on`. You have the local echo of sent CAN frames and the loopback from the CAN controller itself.
This also shows, that the Linux system can talk to the controller in a correct way.
But it is really weird that you get 'garbage data' ...
Can you try without loopback, increase the sjw value to 4 and send the output of `ip -det link show can0`?

mwrouse, 2019-03-25:
@hartkopp How do I increase the sjw value?

Here is `ip -det link show can0`: 
```
3: can0: <NOARP,UP,LOWER_UP,ECHO> mtu 16 qdisc pfifo_fast state UNKNOWN mode DEFAULT group default qlen 10
    link/can  promiscuity 0
    can state ERROR-ACTIVE restart-ms 0
          bitrate 500000 sample-point 0.750
          tq 250 prop-seg 2 phase-seg1 3 phase-seg2 2 sjw 1
          mcp251x: tseg1 3..16 tseg2 2..8 sjw 1..4 brp 1..64 brp-inc 1
          clock 4000000numtxqueues 1 gso_max_size 65536 gso_max_segs 65535
```

That is after fresh boot and running `sudo ip link set can0 up type can bitrate 500000`

hartkopp, 2019-03-25:
`sudo ip link set can0 up type can bitrate 500000 sjw 4`

mwrouse, 2019-03-25:
Then I get:

```
 3: can0: <NOARP,UP,LOWER_UP,ECHO> mtu 16 qdisc pfifo_fast state UNKNOWN mode DEFAULT group default qlen 10
    link/can  promiscuity 0
    can state ERROR-ACTIVE restart-ms 0
          bitrate 500000 sample-point 0.750
          tq 250 prop-seg 2 phase-seg1 3 phase-seg2 2 sjw 2
          mcp251x: tseg1 3..16 tseg2 2..8 sjw 1..4 brp 1..64 brp-inc 1
          clock 4000000numtxqueues 1 gso_max_size 65536 gso_max_segs 65535
```

hartkopp, 2019-03-25:
Ok, and do you see any difference when interacting with the Ardunios? Do you still get the garbage?

mwrouse, 2019-03-25:
Yes, still get the garbage with one Arduino (but it only starts when the MCP2515 on the Arduino starts reporting TX failure). 
Still nothing with both Arduinos on the bus (when TX success starts again).

hartkopp, 2019-03-25:
I assume there's a slight mismatch in the CAN bittiming. Can you get the detailled bittimings from the Ardunio CAN library? The `sample-point 0.750 tq 250 prop-seg 2 phase-seg1 3 phase-seg2 2 sjw 2 and clock' are relevant. You might also try to move the sample-point away from 75%, e.g. by adding `sample-point 0.875` to the `ip`command.

mwrouse, 2019-03-25:
For 500Kbps, and 8MHz crystal, the Arduino CAN Library places these in the CNF registers: 

CNF1(0x2A): 0x00
CNF2(0x29): 0x90
CNF3(0x28): 0x82

I tried changing these to the values I've used in other projects (custom drivers on TI MSP430F5529): 
CNF1: 0x00
CNF2: 0xF8
CNF3: 0x05

But these values also did not work.

hartkopp, 2019-03-25:
Here is how the values are prepared for the mcp251x driver:
https://elixir.bootlin.com/linux/v5.0/source/drivers/net/can/spi/mcp251x.c#L592
Don't know how to switch on the available debug messages right now.
But I think you can only compare the values when having the same clock speed.
Do you know `can-calc-bit-timing mcp251x` ?

mwrouse, 2019-03-25:
I am not familiar with it, saw the file in this repo and that is it. 
Ran the command, changed the Arduino library to write the values reported in that file to the CNF registers, still no luck, used `sample-point 0.875`.

Could it be possible that the values that the drivers are writing are not the same as the ones reported in `can-calc-bit-timing` program?

hartkopp, 2019-03-25:
You can provide the tq / prop-seg / phase-seg1 / phase-seg2 / sjw values directly to the `ip`tool which are then stored into the controller registers without using the build-in algorithm. `calc-can-bittiming` does the calculation with the in-kernel algorithm - and you can provide different clocks too. See `can-calc-bittiming -?` 
When working with `can-calc-bit-timing` I was not able to get the CNF1/2/3 values you provided above. Probably the Ardunio lib has a problem in bittiming calculation - and therefore it only works with each other?!?!

mwrouse, 2019-03-25:
Assuming that the values the Linux drivers writes to the MCP2515 are the ones provided by `can-calc-bit-timing` then just changing the Arduino to write those values to the CNF registers should allow them to match bittiming. 

Which would mean they should be communicating, correct?

hartkopp, 2019-03-25:
Yep. I would give it a try.

mwrouse, 2019-03-26:
I have tried the following CNF1/2/3 values given by: 
```
>  can-calc-bit-timing -b 500000 -c 8000000 mcp251x
Bit timing parameters for mcp251x (cmd-line) with 8.000000 MHz ref clock
nominal                                 real Bitrt   nom  real SampP
Bitrate TQ[ns] PrS PhS1 PhS2 SJW BRP Bitrate Error SampP SampP Error CNF1 CNF2 CNF3
 500000    125   6    7    2   1   1  500000  0.0% 87.5% 87.5%  0.0% 0x00 0xb5 0x01
```
But the issue still persists. I've made sure that the new CNF values were flashed to both Arduinos.

I'll look into another library and open an issue with them. 

If you can get that `dev_dbg` message from the kernel that would be great. Or could point me to how to do it.

Surely I can't be the only person to have wanted to do this, but can't seem to find anything on the issue.

mwrouse, 2019-04-03:
So I'm just giving an update since this (and other threads I created) are now showing up on Google. 

I realized that `ip -det link show can0` was showing the frequency as 4MHz, even though in `/boot/config.txt` it was set to 8MHz. I remember reading on some tutorials that you need to set it to twice the frequency in the config, that now makes sense. Doubling the frequency in the config file made `ip -det` show the proper 8MHz. Unfortunately, that did not solve the problem. 

I created an Arduino sketch to go through all possible combinations of the `CNF1`, `CNF2`, and `CNF3` registers, and attempt to send a message 100 times using each combination. During this, the raspberry pi just remained on `candump can0`. 

Unfortunately, no messages were ever received that had the proper ID, or a length larger than 0.  

I created [this thread on /r/raspberry_pi](https://www.reddit.com/r/raspberry_pi/comments/b87u5q/help_with_can_on_raspberry_pi/), and [this thread on /r/arduino](https://www.reddit.com/r/arduino/comments/b64vii/sending_can_messages_to_raspberry_pi/), both have not been helpful.

At this point, I'm ready to just ditch the Raspberry Pi reading CAN messages and just have an Arduino do it, and send the data to/from the PI over Serial.

hartkopp, 2019-04-04:
That's very sad! I googled about the MCP2515 boards myself and found a potential problem that can be caused by having only 3.3V at the CAN Transceiver (NOT the MCP2515 CAN controller which works fine with 3.3V):
https://www.raspberrypi.org/forums/viewtopic.php?t=141052

hartkopp, 2019-08-24:
No update and not really a can-utils problem. Closing this issue therefore.

closed 2019-08-24 by hartkopp

magicsmoke, 2021-07-23:
Sorry to gravedig.

I've been getting the same issues which is super unfortunate..

From the research that I have done.. It seems like I may need to go down the slcand bridge route as well.

hartkopp, 2021-07-23:
This thread was about the MCP2515 which had many improvements in recent kernels for the RasPi.
If you have a problem with slcand it probably makes sense to create a new issue.

magicsmoke, 2021-07-23:
Hi hartkopp.

Yeah, I've been trying to interface the mcp2515 with the odroid n2+. 

Like OP he revieces packets, but with the wrong ID and no data.

mwrouse, 2021-07-23:
@magicsmoke also sent me an email about this (not the first one I've received about it either), but I'll just respond here so it's all there.

I was never able to get this problem fixed and I haven't tried since April 2019. I figured out a workaround and the project moved on. 

I tested every single combination of configuration in the registry that I could program and scoured the datasheets and source code for any insight. I came up with nothing. All configuration matched with the Raspberry Pi and the Arduino library and I couldn't think of any reason they were not communicating beyond weird electrical stuff.

My solution was to slap an Arduino Mini on the circuit board I was making, have that be responsible for CAN communication. All it did was relayed the raw CAN messages to the Raspberry Pi over serial, and sent CAN messages the Raspberry Pi sent it. 

The Arduino did no processing, just sent and received CAN messages and passed them along over serial.

[Here is a gist with the basic code used on that Arduino.](https://gist.github.com/mwrouse/7fad4bd487c619e11bdd8907686cd847)

We had a semi-custom library for our CAN drivers, so that's where the `CAN` class comes from, and the `can_message` struct. But you can use whatever you want.

hartkopp, 2021-07-23:
Hm, it could be either the level shift or the MCP2515 clock. I remember some weird settings and definitions for clock dividers depending on the provided chip clock. Maybe this initial MCP2515 clock configuration which you already pointed out here: https://github.com/linux-can/can-utils/issues/133#issuecomment-479543935

In fact I have no clue either. Maybe @marckleinebudde has an idea?

Btw. did you

1. Connect CAN_L, CAN_H and GND to both of the Arduino and the RasPi CAN shield?
2. Did you add two parallel 120 Ohm termination resistors between CAN_L and CAN_H?
3. Are one or both nodes built with optical (galvanic) decouplers?

magicsmoke, 2021-07-23:
@mwrouse thanks for the reply.

Yeah this really seems like something to do with the clock speeds of the mcp2515. Regarding the 8Mhz oscillator.

I think I'll have to also build a bridge just using a basic micro and relay the info over. Ive spent way too long than I like trying to get readable data. 

I doubt its actually a hardware issue, my mcp2515 actually works, and revieces packets, it's how the packets are interpreted which is the issue.

[image]
[image]

magicsmoke, 2021-07-23:
> Btw. did you
> 
> 1. Connect CAN_L, CAN_H and GND to both of the Arduino and the RasPi CAN shield?
> 2. Did you add two parallel 120 Ohm termination resistors between CAN_L and CAN_H?
> 3. Are one or both nodes built with optical (galvanic) decouplers?

Yeah I'm able to read the Canbus fine with my oscilloscope, with valid data. No issues with that physical layer. This has to be a software issue.

hartkopp, 2021-07-23:
Have you ever tried to move the sample point to 0.750 (75%) on the nodes?
Which bitrate are you using?

magicsmoke, 2021-07-23:
I have tried setting it to 0.750

Bitrate is 100kbps.

Oddly enough my scope decodes correct data at 50kbps.

hartkopp, 2021-07-23:
Can you scope also measure the crystal of the clock whether it is really 8 MHz?

magicsmoke, 2021-07-23:
I'll be able to measure it tomorrow.

Currently my can network consists of three nodes.

1. ESP32 (built in controller 8Mhz) and transceiver
2. ESP8266 w/ MCP2515 8Mhz module.
3. Odroid-n2+ w/ MCP2515 8Mhz module.

Nodes 1 and 2 can talk to each other. They both have their nitrates set at 100kbps.

I've read somewhere about the actual bitrate halving because of the clock rate.

This is a super weird issue and I'm sure it's just a configuration problem. 

From what I understand if you have two SBC's talk to each other it's fine, but if you introduce any arduino nodes it just can't read it. 

You can clearly see that the data coming from the two arduino framework nodes has the correct packets with no errors.

marckleinebudde, 2021-07-23:
Just as a reminder, the mcp2515 chip has a fixed internal division by 2 of the external clock. If you have an external clock of 8 MHz in `ip -details link show can0` it will show up with a clock of 4 MHz, as this is the effective clock for the bit rate calculation.

magicsmoke, 2021-07-23:
@marckleinebudde

Thanks for that. Yeah I guess the clock is 5Mhz as the spi-max-frequency is 10Mhz in the device tree config.. Right?

```
root@odroid:~# ip -details link show can0
3: can0: <NOARP,UP,LOWER_UP,ECHO> mtu 16 qdisc fq_codel state UP mode DEFAULT group default qlen 10
    link/can  promiscuity 0
    can <LISTEN-ONLY> state ERROR-ACTIVE restart-ms 0
          bitrate 100000 sample-point 0.800
          tq 1000 prop-seg 3 phase-seg1 4 phase-seg2 2 sjw 1
          mcp251x: tseg1 3..16 tseg2 2..8 sjw 1..4 brp 1..64 brp-inc 1
          clock 5000000 numtxqueues 1 numrxqueues 1 gso_max_size 65536 gso_max_segs 65535
```

marckleinebudde, 2021-07-23:
No.

There is the oscillator which drives the mcp2515 and the SPI clock, these are different things. The CAN bit rate is derived from the oscillator, which has an internal divider of 2.

You should have a proper clock node linked to you mcp251x device in the DT (or a `clock-frequency` property). See https://elixir.bootlin.com/linux/v5.13.4/source/arch/arm/boot/dts/imx6q-prti6q.dts#L179 it defines a `can_osc` with 25MHz, the maximum SPI frequency is 5MHz.

magicsmoke, 2021-07-23:
Alrighty, I'll give it a shot later to add a clock-frequency property.

Cheers!

marckleinebudde, 2021-07-23:
Better add a proper `clocks` property.

magicsmoke, 2021-07-25:
Tried making a DTS file for can0, not sure if I did it right.. If I did, it didn't work.

```
/dts-v1/;
/plugin/;

#include <dt-bindings/gpio/meson-g12a-gpio.h>
#include <dt-bindings/gpio/gpio.h>

/ {    
        fragment@0 {
                target = <&spicc0>;

                __overlay__ {
                        status = "okay";
                        can_osc: can-osc {
                                compatible = "fixed-clock";
                                #clock-cells = <0>;
                                clock-frequency = <16000000>;
                        };

                        can0: can@0 {
                                compatible = "microchip,mcp2515";
                                gpio-irq = <&gpio GPIOX_3 GPIO_ACTIVE_LOW>;
                                reg = <1>;
                                clocks = <&can_osc>;
                                spi-max-frequency = <10000000>;
                                status = "okay";
                        };
                };
        };
};

```

marckleinebudde, 2021-07-25:
Have a look at the rpi overlay: https://github.com/raspberrypi/linux/blob/rpi-5.10.y/arch/arm/boot/dts/overlays/mcp251xfd-overlay.dts

The relevant part is:
- You have to adjust the target for the spi
- You probably have to adjust the pintrl - On the raspi the IRQ pin needs to be configured for input.

```
	fragment@0 {
		target = <&gpio>;
		__overlay__ {
			mcp251xfd_pins: mcp251xfd_pins {
				brcm,pins = <25>;
				brcm,function = <BCM2835_FSEL_GPIO_IN>;
			};
		};
	};

	fragment@1 {
		target-path = "/clocks";
		__overlay__ {
			clk_mcp251xfd_osc: mcp251xfd-osc {
				#clock-cells = <0>;
				compatible = "fixed-clock";
				clock-frequency = <40000000>;
			};
		};
	};

	mcp251xfd_frag: fragment@2 {
		target = <&spi0>;
		__overlay__ {
			status = "okay";
			#address-cells = <1>;
			#size-cells = <0>;

			mcp251xfd: mcp251xfd@0 {
				compatible = "microchip,mcp251xfd";
				reg = <0>;
				pinctrl-names = "default";
				pinctrl-0 = <&mcp251xfd_pins>;
				spi-max-frequency = <20000000>;
				interrupt-parent = <&gpio>;
				interrupts = <25 IRQ_TYPE_LEVEL_LOW>;
				clocks = <&clk_mcp251xfd_osc>;
			};
		};
	};
```

> Tried making a DTS file for can0, not sure if I did it right.. If I did, it didn't work.
>
> ```
> /dts-v1/;
> /plugin/;
>
> #include <dt-bindings/gpio/meson-g12a-gpio.h>
> #include <dt-bindings/gpio/gpio.h>
>
> / [...]

You put the clock directly into the SPI node. Quite unusual, might work though.

> 
>                         can0: can@0 {
>                                 compatible = "microchip,mcp2515";
>                                 gpio-irq = <&gpio GPIOX_3 GPIO_ACTIVE_LOW>;

That's not the correct way to specify an interrupt. See above.

>                                 reg = <1>;
>                                 clocks = <&can_osc>;
>                                 spi-max-frequency = <10000000>;
>                                 status = "okay";
>                         };
>                 };
>         };
> };
> ```

magicsmoke, 2021-07-25:
> That's not the correct way to specify an interrupt. See above.

That's at least how the original kernel from Hardkernel does it.

See:
https://github.com/hardkernel/linux/blob/662cd6d8e3e1517d50a56caf6ce19f8d0ef47a0b/arch/arm64/boot/dts/amlogic/overlays/odroidn2/can0.dts#L16

I'll probably have the clocks node in the wrong spot, looking at the rpi example.

marckleinebudde, 2021-07-25:
> That's at least how the original kernel from Hardkernel does it.

Sigh - they have hacked the driver in their vendor kernel :(

https://github.com/hardkernel/linux/commit/15639b4123ccdd316dcbbecd994ef51fc96ae372#diff-55b4d8281f6fd3fe6b7b66da9ae5e9ce818e02b8ac9b23260593fd551c22fdc1

Are you using mainline or that kernel?

magicsmoke, 2021-07-25:
I'm using their flavor kernel.

```
root@odroid:~# cat /proc/version
Linux version 4.9.241-114 (root@builder_n2) (gcc version 7.5.0 (Ubuntu/Linaro 7.5.0-6ubuntu2) ) #1 SMP PREEMPT Tue Apr 20 13:25:36 UTC 2021
```

marckleinebudde, 2021-07-25:
that driver is officially broken:

https://github.com/hardkernel/linux/commit/15639b4123ccdd316dcbbecd994ef51fc96ae372#diff-55b4d8281f6fd3fe6b7b66da9ae5e9ce818e02b8ac9b23260593fd551c22fdc1R1071

```
	clk = devm_clk_get(&spi->dev, NULL);
	if (IS_ERR(clk)) {
￼		if (pdata)
￼			freq = pdata->oscillator_frequency;
￼		else
￼#if defined(CONFIG_ARCH_MESON64_ODROID_COMMON)
￼			freq = spi->max_speed_hz;
```
If there is no clock property, they set the oscillator frequency to the max spi clock speed?
```
￼#else
￼			return PTR_ERR(clk);
￼#endif
￼	} else {
￼		freq = clk_get_rate(clk);
￼	}
```

- and they hacked in the "pinconfig"
- and in deed they are using the `gpio-irq` property?

Why have they done this to the driver? It works out of the box, if you do a proper pinmux setup inthe DT. :disappointed:

magicsmoke, 2021-07-25:
Okay, thanks. Well at least that helps to know.

Ill report this to them, and wishful hoping that they will fix it..

I assume you have no recommendations that could work in the meanwhile without fixing it all myself?

marckleinebudde, 2021-07-25:
> I'm using mainline.
> 
> ```
> root@odroid:~# cat /proc/version
> Linux version 4.9.241-114 (root@builder_n2) (gcc version 7.5.0 (Ubuntu/Linaro 7.5.0-6ubuntu2) ) #1 SMP PREEMPT Tue Apr 20 13:25:36 UTC 2021
> ```

It's quite old, not even v5.10.
That `-114` in the version means it's probably patched,

marckleinebudde, 2021-07-25:
> Okay, thanks. Well at least that helps to know.
> 
> Ill report this to them, and wishful hoping that they will fix it..

"Them" is hardkernel?

> I assume you have no recommendations that could work in the meanwhile without fixing it all myself?

To make it right:
Use a recent mainline kernel, write a proper overlay with proper pinctl and clock setup.

Hack it into existence:
Use the kernel and DT you have been using before, where you were able to receive at least broken CAN frames, add a clock node.

magicsmoke, 2021-07-25:
> "Them" is hardkernel?

Correct. Here is the thread I have with them:

https://forum.odroid.com/viewtopic.php?f=180&t=42897

> To make it right:
> Use a recent mainline kernel, write a proper overlay with proper pinctl and clock setup.
> 
> Hack it into existence:
> Use the kernel and DT you have been using before, where you were able to receive at least broken CAN frames, add a clock node.

Alrighty. This is the DTS i have now hacked. Still isn't working. 

Sorry, This is all new stuff to me, and don't quite understand kernel hacking completely (yet) 😰 Thanks for the patience. I really do appreciate it. 

```
/dts-v1/;
/plugin/;

#include <dt-bindings/gpio/meson-g12a-gpio.h>
#include <dt-bindings/gpio/gpio.h>

/ {     
        fragment@0 {
                target-path = "/";
                __overlay__ {
                        clk_mcp2515_osc: mcp2515-osc {
                                #clock-cells = <0>;
                                compatible = "fixed-clock";
                                clock-frequency = <16000000>;
                        };
                };
        };
        fragment@1 {
                target = <&spicc0>;
                __overlay__ {
                        status = "okay";
                        can0: can@0 {
                                compatible = "microchip,mcp2515";
                                gpio-irq = <&gpio GPIOX_3 GPIO_ACTIVE_LOW>;
                                reg = <1>;
                                clocks = <&clk_mcp2515_osc>;
                                spi-max-frequency = <10000000>;
                                status = "okay";
                        };
                };
        };
};

```

Still receiving frames, but borked data.

 ```
 can0  70F   [0]
  can0  70F   [0]
  can0  70F   [0]
  can0  70F   [0]
  can0  70F   [0]
  can0  70F   [0]
  can0  70F   [0]
  can0  70F   [0]
  can0  70F   [0]
  can0  70F   [0]
  can0  70F   [0]
```

```
root@odroid:~# ip -details link show can0
3: can0: <NOARP,UP,LOWER_UP,ECHO> mtu 16 qdisc fq_codel state UP mode DEFAULT group default qlen 10
    link/can  promiscuity 0
    can state ERROR-PASSIVE restart-ms 0
          bitrate 50000 sample-point 0.750
          tq 1000 prop-seg 7 phase-seg1 7 phase-seg2 5 sjw 1
          mcp251x: tseg1 3..16 tseg2 2..8 sjw 1..4 brp 1..64 brp-inc 1
          clock 5000000 numtxqueues 1 numrxqueues 1 gso_max_size 65536 gso_max_segs 65535

```

marckleinebudde, 2021-07-26:
> I'm using mainline.
> 
> ```
> root@odroid:~# cat /proc/version
> Linux version 4.9.241-114 (root@builder_n2) (gcc version 7.5.0 (Ubuntu/Linaro 7.5.0-6ubuntu2) ) #1 SMP PREEMPT Tue Apr 20 13:25:36 UTC 2021
> ```

`4.9.241-114` is not mainline, it's a tag from the hardkernel repo with the patches mcp2515 driver.

https://github.com/hardkernel/linux/releases/tag/4.9.241-114

marckleinebudde, 2021-07-26:
> ```
> root@odroid:~# ip -details link show can0
> 3: can0: <NOARP,UP,LOWER_UP,ECHO> mtu 16 qdisc fq_codel state UP mode DEFAULT group default qlen 10
>     link/can  promiscuity 0
>     can state ERROR-PASSIVE restart-ms 0
>           bitrate 100000 sample-point 0.750
>           tq 1000 prop-seg 7 phase-seg1 7 phase-seg2 5 sjw 1
>           mcp251x: tseg1 3..16 tseg2 2..8 sjw 1..4 brp 1..64 brp-inc 1
>           clock 5000000 numtxqueues 1 numrxqueues 1 gso_max_size 65536 gso_max_segs 65535

It still uses half of `spi->max_speed_hz`.

> ```

Can you post the output of `cat /sys/kernel/debug/clk/clk_summary` and `dtc -i fs /proc/device-tree`?

marckleinebudde, 2021-07-26:
> https://forum.odroid.com/viewtopic.php?f=180&t=42897
>> It seems not to easy to find which part of the driver was broken.

All changes they did to the driver are not needed, there are proper interfaces to do this via the DT.
They don't understand the meaning of `spi->max_speed_hz`. That's the maximum speed of the SPI bus and has nothing to do with the external oscillator of the mcp2515.

magicsmoke, 2021-07-28:
Thanks for all your efforts @marckleinebudde. I _really_ appreciate it.

My .dts would have probably worked if I put it in the right place after compiling..

Cap00k on the hardkernel forum posted [this fix](https://forum.odroid.com/viewtopic.php?p=332651#p332651) and I managed to get it to work 🥳

```
root@odroid:~# ip -details link show can0
3: can0: <NOARP,UP,LOWER_UP,ECHO> mtu 16 qdisc fq_codel state UP mode DEFAULT group default qlen 10
    link/can  promiscuity 0
    can <TRIPLE-SAMPLING> state ERROR-PASSIVE restart-ms 0
          bitrate 100000 sample-point 0.875
          tq 625 prop-seg 6 phase-seg1 7 phase-seg2 2 sjw 1
          mcp251x: tseg1 3..16 tseg2 2..8 sjw 1..4 brp 1..64 brp-inc 1
          clock 8000000 numtxqueues 1 numrxqueues 1 gso_max_size 65536 gso_max_segs 65535

root@odroid:~# candump any
  can0  1F4   [2]  03 00
  can0  1F4   [2]  04 00
  can0  1F4   [2]  05 00
  can0  1F4   [2]  06 00
  can0  1F4   [2]  00 01
  can0  1F6   [5]  00 00 00 00 00
  can0  1F4   [2]  01 01
  can0  1F4   [2]  02 01
  can0  1F6   [5]  01 00 00 00 00
  can0  1F4   [2]  03 01
  can0  1F6   [5]  02 00 00 00 00
  can0  1F4   [2]  04 01
  can0  1F4   [2]  05 01
  can0  1F4   [2]  06 01
  can0  1F4   [2]  00 00
  can0  1F4   [2]  01 00
  can0  1F4   [2]  02 00
  can0  1F4   [2]  03 00
  can0  1F6   [5]  00 00 00 00 00
  can0  1F4   [2]  04 00
  can0  1F6   [5]  01 00 00 00 00
  can0  1F4   [2]  05 00
  can0  1F4   [2]  06 00
  can0  1F6   [5]  02 00 00 00 00
  can0  1F4   [2]  00 01
  can0  1F4   [2]  01 01

```

But yeah, the underlying issue is the broken driver, like you said. 

Thanks again though to everyone who has helped, you've been _**really**_ helpful.

marckleinebudde, 2021-07-28:
What was the difference to the overlay I posed?

mwrouse, 2021-07-28:
So how can other people seeing this thread in the future solve this problem? I am not familiar with Linux enough to follow what is going on here. 

Are these files something anyone with Raspbian can change to solve this problem?

marckleinebudde, 2021-07-29:
Hey @mwrouse

There is no problem with Raspbian/Rasperry PI OS, the problem is that hardkernel did some changes to the driver...The proper solution would be to use a mainline kernel. The modified driver seems to work with this DT overlay:

```
/dts-v1/;
/plugin/;

#include <dt-bindings/gpio/meson-g12a-gpio.h>
#include <dt-bindings/gpio/gpio.h>

/ {
	fragment@0 {
		target-path = "/";
		__overlay__ {
			can_clk: can-clk {
				compatible = "fixed-clock";
				clock-frequency = <24000000>;
				clock-output-names = "can_clk";
				#clock-cells = <0>;
			};
		};
	};

	fragment@1 {
		target = <&spicc0>;

		__overlay__ {
			status = "okay";

			can0: can@0 {
				compatible = "microchip,mcp2515";
				reg = <1>;
				clocks = <&can_clk>;
				gpio-irq = <&gpio GPIOX_3 GPIO_ACTIVE_LOW>;
				spi-max-frequency = <10000000>;
				status = "okay";
			};
		};
	};
};
```

mwrouse, 2021-07-29:
I originally experienced the problem on a raspberry pi running Raspbian. So this fix is for a different OS.

Is it possible the problem I was having on Raspbian is a similar issue?

I am obviously aware that this is not the forum for this, but people will find this in the future, so we might as well try and give enough info, point them somewhere they can get help, or move this forward and get it fixed somewhere else.

marckleinebudde, 2021-07-30:
Is your mcp2515 working on the raspi?

dumtux, 2022-11-13:
@marckleinebudde Did you find a solution?
I am having similar trouble on Omega2.

marckleinebudde, 2022-11-13:
Feel free to open an issue and explain your problem.

dumtux, 2022-11-16:
@marckleinebudde 

I am customizing OpenWrt 22.03 (kernel 5.10) for my Omega2 hardware.
I connected MCP2515 to the SPI bus, on CS1 (GPIO11).
The DTS file looks like this.

```
&spi0 {
	status = "okay";

	pinctrl-names = "default";
	pinctrl-0 = <&spi_pins>, <&spi_cs1_pins>;

	flash0: flash@0 {
		compatible = "jedec,spi-nor";
		reg = <0>;
		spi-max-frequency = <40000000>;
		...
	};

	can0: can@1 {
		compatible = "microchip,mcp2515";
		reg = <1>;
		spi-max-frequency = <5000000>;
		clock-frequency  =  <20000000>;
		interrupt-parent = <&gpio>;
		interrupts = <11 IRQ_TYPE_EDGE_FALLING>;
	};
};
```

On MCP251x kernel module loading, `dmesg` says it  is initilaized succesfully.
I can enable `can0` and send messages using `cansend` command.
But I cannot get `candump` working.
If I connect external CAN device to send some message to the board, the INT pin falls to LOW, that means it is getting into some inturrupt status.

But the `candump` can read nothing. And from then, I cannnot make `cansend` working again.
I have to restart `can0` to make `cansend` working again.

When I connect two same board, and enable `can0` on board A then enable `can0` on board B, both boards can not send using `cansend`, just like the above. INT pins of both board falls to LOW and doesn't recover.

Here's detailed discussion on [Onion Omega2 Community](https://community.onion.io/topic/4892/can-bus-using-mcp2515-with-omega2).
There are what I've tried until now, e.g. using `listen-only` options, changing INT pin GPIO input directions, etc.

marckleinebudde, 2022-11-16:
Please use the issue I just created.

marckleinebudde, 2025-11-04:
> actually i use 8mhz mcp2515 in stm32f103c8t6 and realtek controller both fixed 500kbs but its reciver side geting data

Thanks for moving this to https://github.com/linux-can/can-utils/issues/603.