linux-can/can-utils #148: isotpsend -> isotprecv unreliable?

brandonros, 2019-07-30:
```
$ echo "36 01 60 00 00 00 00 e0 0f 00 00 00 00 00 fc ff 0f 84 20 22 00 84 00 00 00 00 02 00 31 30 53 57 30 32 38 36 30 31 af af af af af af af af 03 00 00 00 6c 46 79 0d 60 04 00 00 00 20 00 84 fb ff 0f 84 fe ca de fa fe af fe ca 74 ff 37 84 70 ff 0f 84 00 10 00 00 61 04 04 00 1c 37 00 84 5f 47 00 84 fe ca de fa fe af fe ca 00 00 00 00 00 00 00 00 10 10 00 00 62 04 10 00 98 20 00 84 03 22 00 84 fe ca de fa fe af fe ca 00 00 00 00 00 00 00 00 00 10 00 00 cf 58 16 a3 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00" | isotpsend can0 -p AA:AA -s 7e8 -d 7e0
```

```
$ stdbuf -i0 -o0 -e0 isotprecv can0 -l -s 7e0 -d 7e8 -p AA:AA
```

I see the traffic in candump. I see isotprecv correctly sending the control flow to get all of the data, but I need to call the isotpsend call multiple times to get isotprecv to receive the data.

What flags can I change to make this transmission more reliable? All smaller isotpsend + isotprecv calls on this interface work flawlessly and I don't see any error frames/dropped data.

brandonros, 2019-07-30:
`isotpdump` shows the full transmission as successful, but `isotprecv` is still blocking as if it didn't receive anything.

```
$ isotpdump -s 7e0 -d 7e8 can0
 can0  7E8  [8]  [FF] ln: 242  data: 36 01 60 00 00 00 
 can0  7E0  [8]  [FC] FC: 0 = CTS # BS: 0 = off # STmin: 0x00 = 0 ms
 can0  7E8  [8]  [CF] sn: 1    data: 00 E0 0F 00 00 00 00 
 can0  7E8  [8]  [CF] sn: 2    data: 00 FC FF 0F 84 20 22 
 can0  7E8  [8]  [CF] sn: 3    data: 00 84 00 00 00 00 02 
 can0  7E8  [8]  [CF] sn: 4    data: 00 31 30 53 57 30 32 
 can0  7E8  [8]  [CF] sn: 5    data: 38 36 30 31 AF AF AF 
 can0  7E8  [8]  [CF] sn: 6    data: AF AF AF AF AF 03 00 
 can0  7E8  [8]  [CF] sn: 7    data: 00 00 6C 46 79 0D 60 
 can0  7E8  [8]  [CF] sn: 8    data: 04 00 00 00 20 00 84 
 can0  7E8  [8]  [CF] sn: 9    data: FB FF 0F 84 FE CA DE 
 can0  7E8  [8]  [CF] sn: A    data: FA FE AF FE CA 74 FF 
 can0  7E8  [8]  [CF] sn: B    data: 37 84 70 FF 0F 84 00 
 can0  7E8  [8]  [CF] sn: C    data: 10 00 00 61 04 04 00 
 can0  7E8  [8]  [CF] sn: D    data: 1C 37 00 84 5F 47 00 
 can0  7E8  [8]  [CF] sn: E    data: 84 FE CA DE FA FE AF 
 can0  7E8  [8]  [CF] sn: F    data: FE CA 00 00 00 00 00 
 can0  7E8  [8]  [CF] sn: 0    data: 00 00 00 10 10 00 00 
 can0  7E8  [8]  [CF] sn: 1    data: 62 04 10 00 98 20 00 
 can0  7E8  [8]  [CF] sn: 2    data: 84 03 22 00 84 FE CA 
 can0  7E8  [8]  [CF] sn: 3    data: DE FA FE AF FE CA 00 
 can0  7E8  [8]  [CF] sn: 4    data: 00 00 00 00 00 00 00 
```

brandonros, 2019-07-30:
```
$ lsmod | grep can
can_raw                20480  1
can_isotp              20480  1
can                    49152  2 can_isotp,can_raw
can_dev                24576  1 gs_usb
```

brandonros, 2019-07-30:
```
~$ sudo modinfo can_isotp
filename:       /lib/modules/4.9.0-9-amd64/extra/net/can/can-isotp.ko
alias:          can-proto-6
author:         Oliver Hartkopp <socketcan@hartkopp.net>
license:        Dual BSD/GPL
description:    PF_CAN isotp 15765-2:2016 protocol
depends:        can
retpoline:      Y
vermagic:       4.9.0-9-amd64 SMP mod_unload modversions 
```

hartkopp, 2019-07-30:
> `isotpdump` shows the full transmission as successful, but `isotprecv` is still blocking as if it didn't receive anything.

This wasn't the full transmission. Try it with a virtual CAN interface (e.g. vcan0) and you can see more than 20 frames.

Have you connected your CAN adapter so a second CAN node that acknowledges the sent frames? A hardware CAN adapter can not work alone on the CAN bus.

brandonros, 2019-07-30:
vcan0 fixed my issue. Thank you, I did not know about the acknowledgement of sent frames underlying.

For anybody else, it was

```
$ sudo modprobe vcan
$ sudo ip link add dev vcan0 type vcan
$ sudo ip link set up vcan0
```

`stdbuf -i0 -o0 -e0 isotprecv vcan0 -l -s 7e0 -d 7e8 -p AA:AA | node uds-server.js`
`stdbuf -i0 -o0 -e0 isotprecv vcan0 -l -s 7e8 -d 7e0 -p AA:AA | node uds-client.js`

do I have those source + destination IDs correct?

hartkopp, 2019-07-30:
> vcan0 fixed my issue. Thank you, I did not know about the acknowledgement of sent frames underlying.

The issue does not show up immediately as the CAN drivers have some TX queue len (usually 10 frames) which are filled - and then the CAN controller is blocked and waiting for the ACK.

> `stdbuf -i0 -o0 -e0 isotprecv vcan0 -l -s 7e0 -d 7e8 -p AA:AA | node uds-server.js`
> `stdbuf -i0 -o0 -e0 isotprecv vcan0 -l -s 7e8 -d 7e0 -p AA:AA | node uds-client.js`
> 
> do I have those source + destination IDs correct?

Yes. You have a "virtual" point-to-point connection which is defined by two CAN IDs. Therefore the two endpoints receive and transmit on vice versa identifiers.

closed 2019-07-30 by brandonros