linux-can/can-utils #153: Bug - CAN id's are missing

srigandeswaran, 2019-09-19:
Hi all,
  I used these CAN driver code in my code in i.mx6 board interfacing with the infotainment display(radio module), I receive all the CAN commands from infotainment display(radio module) but sometimes,some of the CAN id's from the radio module is missing.can anyone please take a look at that and fix that bug.

hartkopp, 2019-09-19:
Huh ?
No idea what CAN driver you are talking about and why (which) CAN-IDs are missing and what code you are using to access the CAN bus. This is probably an i.MX6 issue but obviously no can-utils issue.
Please contact the Linux CAN mailing list with a detailed bug report.
Thanks.

closed 2019-09-19 by hartkopp

srigandeswaran, 2019-12-09:
@hartkopp I used the candump.c (which is the code to receive the CAN id's from CAN bus),I receive all the CAN id's from the bus,but sometimes some of the CAN id's are missing,that's what I said. ok I will contact the Linux can mailing listing for further things.