linux-can/can-utils #156: Question: configure can bitrate, canconfig & libsocketcan

iurly, 2019-10-13:
Hi,

I was looking for a way of letting a userspace program bring up/down a socketcan interface with a given bitrate (other than `sudo ip link`).
Rationale: `ip link` requires setuid privileges or CAP_NET_ADMIN capabilities, and I could not find an easy way to grant such capabilities to a single user or script.

I came across this old [canconfig.c](https://github.com/jmore-reachtech/can-utils/blob/master/src/canconfig.c) from the original pengutronix code and I was wondering why it's not part of this official package.
I can actually see some use for it. To me, it looks like having a dedicated `canconfig` executable -- over which you can run `sudo setcap 'CAP_NET_ADMIN=eip' ` -- would solve the issue.

@marckleinebudde, it looks like you've been involved with socketcan from its very inception, so I would really appreciate your feedback on this topic.

Thanks!

snickl, 2019-10-28:
Another use case that would be facilitated by the inclusion of canconfig.c: The "ip" program from busybox doesn't support CAN, and switching from the busybox applet to the iproute package just for CAN support may be undesirable for small embedded systems.

mgsanava, 2023-09-13:
@marckleinebudde 
We have a code that makes us of canconfig command. It used to work flawlessly.
Yesterday, we ran that code again after a long time and now it says "canconfig command not found".
I guess that can-utils package was upgraded in our yocto setup which is causing this issue. I'm not sure though.
We are at the latest version 2023.03.
To which version do we need to downgrade to get canconfig to work?
Thanks!

marckleinebudde, 2023-09-13:
@mgsanava `canconfig` was never part if this repository, but it's part of [canutils](https://git.pengutronix.de/cgit/tools/canutils/). You can either install `canutils` or use the `ip` command (of `iproute2`) to configure the bitrate.

mgsanava, 2023-09-13:
That was super fast. Thanks for the details. And, yeah that solves my problem.

marckleinebudde, 2023-09-13:
@snickl There's a patch that adds CAN support to busybox's ip command: http://lists.busybox.net/pipermail/busybox/2023-September/090450.html