linux-can/can-utils #157: J1939 Transport Protocol Receive (Testing with TestJ1939)

BalkanRising, 2019-10-21:
Hello,

I am trying to create a J1939 socket (via C++ using kernel headers) that is capable of receiving J1939 transport protocol messages but have no such luck. I can set up a socket that receives data when I issue the following command:

`testj1939 -s8 vcan0:0x79,0x1200 :0x69`

But receives nothing when I issue the following:

`testj1939 -s9 vcan0:0x79,0x1200 :0x69`

By receives nothing, I mean the call to `recv `returns a -1 where `errno `is 11.

I tried looking at the utilities in this project for examples on how to receive TP messages but I did not find any. I studied the J1939 driver and it seems like what I am doing should work.

**Does anyone have any ideas on how I can set up a socket to receive TP data?**

My code looks like this:

```
int main(int argc, char* argv[]) {
  // Set up rx
  struct sockaddr_can SockAddrRx {};
  SockAddrRx.can_family = AF_CAN;
  SockAddrRx.can_ifindex = if_nametoindex("vcan0");
  SockAddrRx.can_addr.j1939.name = J1939_NO_NAME;
  SockAddrRx.can_addr.j1939.pgn = J1939_NO_PGN;
  SockAddrRx.can_addr.j1939.addr = 0x69;

  // Set up tx
  struct sockaddr_can SockAddrTx {};
  SockAddrTx.can_family = AF_CAN;
  SockAddrTx.can_ifindex = if_nametoindex("vcan0");
  SockAddrTx.can_addr.j1939.name = J1939_NO_NAME;
  SockAddrTx.can_addr.j1939.pgn = J1939_NO_PGN;
  SockAddrTx.can_addr.j1939.addr = 0x79;

  int file_desc = socket(PF_CAN, SOCK_DGRAM, CAN_J1939);
  bind(file_desc, reinterpret_cast<sockaddr *>(&SockAddrRx), sizeof(SockAddrRx));
  connect(file_desc, reinterpret_cast<sockaddr *>(&SockAddrTx), sizeof(SockAddrTx));

  while (1) {
    int ret = recv(file_desc, data_to_rcv, sizeof(data_to_rcv), MSG_DONTWAIT);
    std::cout << std::dec << ret << std::endl;
    std::cout << std::dec << errno << std::endl;
  }
}
```

marckleinebudde, 2019-10-22:
Which kernel are you using?

olerem, 2019-10-22:
Hi, i did following test with latest kernel:
```
root@DistroKit:~ jcat -r vcan0:0x69 > /tmp/ff &
root@DistroKit:~ testj1939 -s8 vcan0:0x79,0x1200 :0x69
root@DistroKit:~ fg
jcat -r vcan0:0x69 1>/tmp/ff

root@DistroKit:~ hexdump /tmp/ff
0000000 2301 6745 ab89 efcd
0000008
root@DistroKit:~ jcat -r vcan0:0x69 > /tmp/ff &
root@DistroKit:~ testj1939 -s9 vcan0:0x79,0x1200 :0x69
root@DistroKit:~ fg
jcat -r vcan0:0x69 1>/tmp/ff

root@DistroKit:~ hexdump /tmp/ff
0000000 2301 6745 ab89 efcd 0001
0000009
```
So far, it seems to work as expected.

Error 11 means: try again, there is no data.
Is it working without MSG_DONTWAIT?

If it is not working, please paste some compilable code here.

BalkanRising, 2019-10-22:
Oleksij, Marc, 

I ran a version of the test you describe before posting this issue and it did not work when sending -s9. I tried again right now, exactly as you have written, and I get data with -s8 but not with -s9.

My kernel version is 4.9.9 with the following version of the J1939 drivers: https://github.com/kurt-vd/linux/tree/j1939d-v4.9.10. This means that I am not using the latest version of can-utils because the latest version of jcat uses SO_J1939_ERRQUEUE and my version of the J1939 driver does not have that implemented. I am using the following commit of can-utils to do my testing (The latest commit before ERRQUEUE support was added): `05368ef1aebf9d339afbd083d3f5671ae3fba3a2 `

The version of the J1939 drivers that I have seem to support TP and ETP so, while I acknowledge that the drivers are out of date, it seems to me that they should work. I was hoping to be able to get this working without updating the kernel/drivers.

marckleinebudde, 2019-10-22:
Anything older than v5.4-rc1 is probably not going to work. We've changed the use space API.

BalkanRising, 2019-10-22:
Sounds good, I will work on updating to the latest drivers then. Appreciate the help!

closed 2019-10-22 by BalkanRising